Bouni/kicad-jlcpcb-tools #145: No schematics generated

thote1276, 2022-03-08:
HI, I test on a basic project to generate JLCPCB files. I succesfully success to have my gerber but after chossing all my component, I don't have any files in assembly.

Bouni, 2022-03-09:
Do you get any output in the message box at the bottom when you click the generate fabrication files button?

thote1276, 2022-03-09:
Hi,
In get output for gerber only. Non mentionof assembly as in your small video

Bouni, 2022-03-09:
Can you share your board file so I can test it?

thote1276, 2022-03-09:
I invite you on a repo on github, so that you can see what I get from your plugin. 
I just installed your plugin with the url, open the layout, click on the button of the plugin. 
Then I update the library, choose the component I found on JLC's library, some are missing but clicking on Make production files, i get only the gerber.

Bouni, 2022-03-09:
I've just downloaded your repo, opend the project and started the plugin. then clicked generate fabrication files and it worked 🤔 

[image]

[image]

thote1276, 2022-03-09:
Oh ...
Is there anything i need to download additionnal to your plugin ? Is this better to download the zip and then  paste it in the plugin folder ?

Bouni, 2022-03-09:
Ideally you install it via PCM as explained here: https://github.com/Bouni/kicad-jlcpcb-tools#installation-

thote1276, 2022-03-09:
I tried too.
Nothing more is needed ?
I think I add problem to export a pos file in csv. Can this be related ? A problem on my setup with csv ?

Télécharger Outlook pour Android<https://aka.ms/AAb9ysg>

Bouni, 2022-03-09:
No, if you click the button everything should be done automatically ....

Can you post a picture of the folder structure you get with the files

thote1276, 2022-03-09:
Hi, 
On my side i tried reinstalling Kicad and your library using PCM.

thote1276, 2022-03-09:
I'm stuck there 
[image]

thote1276, 2022-03-09:
Maybe which Kicad are you using ? 6.0.0 ?

Bouni, 2022-03-10:
From your screenshot I see that you're not runnig a version installed via PCM, it says [ dev ] in the bar at the top.
I run 6.0.2 and 6.99 and it works in both for me ...

Please open the console 

[image]

before you start the plugin, then try to generate the files again. Post the output you get in the console output window.
Maybe that helps me figuring out whats going on here.

thote1276, 2022-03-11:
Hi, it seems to work on one of my computer with the console open. 
Today I got the error, database is locked on the first computer.

thote1276, 2022-03-13:
Hi, 
I think the final word was that the antivirus software was blocking one of your script I think !

closed 2022-03-13 by thote1276