Bouni/kicad-jlcpcb-tools #155: Gerbers from Kikit panel

Gio-dot, 2022-04-12:
Hi, thank you for your great work!

I'm new with Kikad world. I realized a panel using kikit, then i made productions files using your plugin. It seems that v-cuts are missing in gerber files. There is something i'm missing? How v-cuts are reported in gerber files?

Bouni, 2022-04-12:
To be honest I never used V-cuts before myself and have no idea how they are included in gerbers 🤔 
Let me see If I can figure that out!

Gio-dot, 2022-04-12:
> To be honest I never used V-cuts before myself and have no idea how they are included in gerbers 🤔 Let me see If I can figure that out!

I don't know. I made a panel with Kikit and i can **only** see V-cut markers on `User.Comments` layer. I think i should there how v-cuts are managed. Thank you
[image]