Bouni/kicad-jlcpcb-tools #156: Discrepancy between settings used and JLCPCB FAQ

FredM67, 2022-04-12:
On this page https://support.jlcpcb.com/article/194-how-to-generate-gerber-and-drill-files-in-kicad-6, the needed and/or recommended settings are shown.
It looks like what the script fabrication.py is using does not fully match the settings from the JLCPCB FAQ !
For example, `popt.SetSubtractMaskFromSilk(False)` while JLCPCB recommands `true`.
Or `popt.SetUseGerberProtelExtensions(False)` where JLCPB recommands `true`.

Bouni, 2022-04-13:
I guess you're right. I'll have to look into the linked JLC article and adjust the options ...

FredM67, 2022-04-13:
Well, I don't want to be right or not :D
At least the stuff with the "ProtelExtension" is strange, because KiCad says "No longer recommended" !

I didn't check the other points about drilling & Co.
Hope, the PCB I've ordered will be right !

Bouni, 2022-04-13:
As KiCAD disencourages the use of Protel Extension I think its not a good idea to set `popt.SetUseGerberProtelExtensions` to true.
They might deprecate it someday in the near future and the the plugin breaks because of that.
I've ordered dozens of PCBs generated with the current settings an all of them came out just fine!

`popt.SetSubtractMaskFromSilk(True)`on the other hand seems like a good thing.

This is generated with `False`

[image]

And this with `True`

[image]

I think I'll set that one to True!

But before I do that I'll check all the other option (some are quite hard to match to a checkbox in the KiCAD UI ...)

FredM67, 2022-04-13:
I agree, this "Protel" extension is out-dated and probably, JLCPCB will change over the new, or they've already done it but forgot to update the FAQ.

Since all your PCBs came just fine, don't lose time with search other option. The one with `SetSubtractMaskFromSilk` is a good idea to change... but leave the rest as it is.

"Never change a running system" ;-)

Bouni, 2022-04-13:
You're right 😄 

Will do that in my lunch break ...

PR #158 (2022-04-13): fix for #165 and #157
- Removed `popt.SetExcludeEdgeLayer(True)` to fix #157
- Changed `popt.SetSubtractMaskFromSilk(False)` to `popt.SetSubtractMaskFromSilk(True)` to fix #156
closed 2022-04-13 by Bouni