Bouni/kicad-jlcpcb-tools #157: No files are generated in the latest nightly 6.99.0-1615-g403992a6f4

Bouni, 2022-04-13:
**Describe the bug**
When the `generate fabrication files` button is clicked, nothing happens.

**To Reproduce**
Steps to reproduce the behavior:
1. Click generate fabrication files
2. Nothing happens

**Expected behavior**
Gerbers, BOM and POS should be generated.

# KiCad Version

```
Application: KiCad PCB Editor (64-bit)

Version: (6.99.0-1615-g403992a6f4), release build

Libraries:
	wxWidgets 3.1.5
	FreeType 2.11.1
	HarfBuzz 2.8.2
	FontConfig 2.13.1
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Apr 11 2022 07:20:34
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.6.0
	Curl: 7.78.0-DEV
	ngspice: 36
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Bouni, 2022-04-13:
The problem seems to be in this line: https://github.com/Bouni/kicad-jlcpcb-tools/blob/2022.04.02/fabrication.py#L117

When the stdout/stderr window of KiCAD is open the problem gets clear:

```
line 117, in generate_geber popt.SetExcludeEdgeLayer(True)
AttributeError: 'PCB_PLOT_PARAMS' object has no attribute 'SetExcludeEdgeLayer'
```

My best guess is that `SetExcludeEdgeLayer` has been removed or renamed in the latest version of KiCAD

Robyrob, 2022-08-09:
i saw that kicad's developers has changed the plot options in
https://gitlab.com/kicad/code/kicad/-/commit/0bdbfe1446a80393f9251fe04c630190c936cc3c
As it was important for me to include edges in gerber file (using CopperCAM), do you think will be  possible to restore this functionality whitn new option SetPlotOnAllLayersSelection ?
Many thenks

Sync-, 2022-12-02:
https://gitlab.com/kicad/code/kicad/-/commit/31bad5baaad6162dfb6cf1c519f28519665d6174
I think this commit is breaking this line:
https://github.com/Bouni/kicad-jlcpcb-tools/blob/2022.04.02/fabrication.py#L117

I cannot generate files in the latest nightly.

Sync-, 2022-12-16:
It's been fixed in a recent Kicad commit! :)

PR #158 (2022-04-13): fix for #165 and #157
- Removed `popt.SetExcludeEdgeLayer(True)` to fix #157
- Changed `popt.SetSubtractMaskFromSilk(False)` to `popt.SetSubtractMaskFromSilk(True)` to fix #156
closed 2022-12-30 by Bouni