Bouni/kicad-jlcpcb-tools #159: Reversed rotation angle with bottom components

t123yh, 2022-04-16:
**Describe the bug**
When component is placed in the bottom layer of the board, rotation angle 0 degree and 180 degree is reversed, causing reversed placement.

**To Reproduce**
The following test is done with the latest kicad-jlcpcb-tools, kicad 6.0.4 and Chinese version of JLCPCB.

This is my footprint:
[image]

This is the PCB in KiCad (apart from the topmost resistor, the other 4 components are all C239773: ESD5341N-2/TR). The positive (+) pins of the ESD5341s are connected to GND (the polygon pour), and negative (-) pins are connected to respective signals.

[image]

This is ESD5341 in [szlcsc.com](https://item.szlcsc.com/238838.html)
[image]

When I export POS with kicad-jlcpcb-tools, D6 and D19's rotate value is 180 degrees, D22 and D23's rotate value is 90 degrees.

[image]

However, when I import them to JLCPCB (the following screenshot is captured from JLCPCB order making), bad thing happens: polarity of D19 and D6 is reversed (note that negative pin is connected to ground instead of positive pin), while D22 and D23 is still correct:

[image]

Maybe 180 deg and 0 deg should be reversed when exporting bottom components?

**Expected behavior**
Please see above.

# KiCad Version

```
Application: KiCad PCB 编辑器

Version: (6.0.4-0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.79.1 SecureTransport (LibreSSL/3.3.5) zlib/1.2.11 nghttp2/1.45.1

Platform: macOS Version 12.3.1 (Build 21E258), 64 bit, Little endian, wxMac

Build Info:
	Date: Mar 17 2022 21:21:05
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.78.0
	OCC: 7.6.1
	Curl: 7.64.1
	ngspice: 35
	Compiler: Clang 12.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

Bouni, 2022-04-23:
Hey, as soon as I'm done with my COVID infection I'll take a look at this, sorry for the delay ...

t123yh, 2022-04-23:
> Hey, as soon as I'm done with my COVID infection I'll take a look at this, sorry for the delay ...

Hi Bouni, no need to apologize, and I'm not in a hurry. I really appreciate your effort and great work. Please take a good rest, I hope you recover soon.

jcb83, 2022-10-20:
I recently stumbled across this issue while trying to rotate some diagonal diodes correctly, which I could not seem to get right no matter what rotation correction I used in the plugin.
This got me investigating, and to cut a long story short: It turns out that the angles of parts on the bottom side are mirrored along the vertical axis in KiCad, but JLCPCB expects them to stay the same. The fix is to mirror the rotation of bottom side parts along the vertical axis, a `rotation = 180 - rotation` did the trick for me.

## A bit more detail
I tested this with some diodes, as they have a recognizable direction in KiCad and on the JLCPCB web preview.

A diode pointing "down" (input side up, marked "output" side down) has a 90° angle on the top as well as the bottom side, and results in the correct placement of the part on the JLCPCB website. 

A diode pointing "left to right" (input left, output right), however, has an angle of 180° on the top side, but 0° on the bottom in KiCad (see screenshots below). JLCPCB will place that part incorrectly on the bottom side, as it seems to expect the angles to be 180° in both cases. Adding a flat 180° rotation correction in the plugin will not fix the issue, as then the parts pointing along the vertical axis will be placed incorrectly (and it does nothing to fix 45° rotated parts).

Screenshots of my test setup, taken from the KiCad 3DViewer. The text next to a part gives the rotation as seen in the KiCad "Properties" window of that part.
[image] [image]

Screenshot of the preview on the JLCPCB website. Note that vertical parts appear correctly, horizontal parts are 180° flipped, and diagonal parts appear 90° rotated in different directions.
[image]

Bouni, 2022-10-24:
@jcb83 Please submit a PR so that I can merge your change! Thank you for your detailed explanation though 👍🏽

closed 2022-12-30 by Bouni