Bouni/kicad-jlcpcb-tools #160: Enable to add custom layers (v-score for example)

AntonKrug, 2022-04-17:
JLPCB allows custom panelization which can be done on the user's side. They allow users to provide custom layer, for example vscore and they grove it out when given instructions to do so. However there is no way in this plugin to let consider (generate such layers)

**Describe the solution you'd like**
Have a 'manage layer' button which would could have checkboxes which layers would be considered. KiKit panelization plugin by default outputs of the v-cut information into the 'User.Comments' layer, this could be the default layer to be considered.

Bouni, 2022-04-26:
Is a custom selection actually needed or would it be sufficiant to just output the `User.Comments` by default?
A additional exported layer would not hurt even if its empty, right?

Bouni, 2022-04-26:
Resolved by #161

closed 2022-04-26 by Bouni