Bouni/kicad-jlcpcb-tools #162: No parts visible in the jlcpcb part placement preview with the generated files.

TT-392, 2022-04-19:
**Describe the bug**
The generated files don't seem to be understood by the jlcpcb review parts placement preview thingey. I tried using this plugin with my own design first, and, even though there are "selected parts", none of them are showing up in the preview. I heard that the kicad coordinate system is a little different than most. For this reason I tried creating a pcb of 30x30cm with the center at the 0,0 coordinate and a bunch of parts spread out over this pcb. But even this pcb didn't have any parts on it (it is not that they weren't in the right place. They just aren't there). I tried creating some test files with easyeda, and those seem to work just fine, so it is not my browser or something like that.

This is what it looks like:
[image]

**To Reproduce**
Use the plugin to create the following files:
[GERBER-JLC pca test.zip](https://github.com/Bouni/kicad-jlcpcb-tools/files/8514234/GERBER-JLC.pca.test.zip)
[BOM-JLC pca test.csv](https://github.com/Bouni/kicad-jlcpcb-tools/files/8514237/BOM-JLC.pca.test.csv)
[POS-JLC pca test.csv](https://github.com/Bouni/kicad-jlcpcb-tools/files/8514238/POS-JLC.pca.test.csv)
Go through the jlcpcb ordering process, and see no parts in the preview.

**Expected behavior**
Visible parts in the preview

# KiCad Version

```
Application: KiCad PCB Editor
Version: (6.0.4), release build
Libraries:
    wxWidgets 3.0.5
    libcurl/7.82.0 OpenSSL/1.1.1n zlib/1.2.12 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.2 libpsl/0.21.1 (+libidn2/2.3.0) libssh2/1.10.0 nghttp2/1.47.0
Platform: Linux 5.15.32-1-lts x86_64, 64 bit, Little endian, wxGTK, i3, x11
Build Info:
    Date: Mar 18 2022 08:05:28
    wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.78.0
    OCC: 7.5.3
    Curl: 7.82.0
    ngspice: 36
    Compiler: GCC 11.2.0 with C++ ABI 1016
Build settings:
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON
```

Bouni, 2022-04-26:
I'll look into this soon

Bouni, 2022-05-17:
Hey, sorry for the delay 😓 

I just did a quick test with a 100x100 pcb and 5 74HC595 on it:

[image]

That worked for me (aside from the offset in the parts displayed).

Does it still not work for you?

TT-392, 2022-05-25:
Yep, still not working for me. Just tried making a new pcb after installing 2022.05.01 of the plugin. I might try it on a windows dual boot to see if it is maybe something with my pc not running the javascript of the site correctly or something? thought that would be kinda strange considering the fact that I tried multiple browser already.

Bouni, 2022-05-30:
Just as an information on this matter: The preview and orientation of the parts is completely irrelevant for JLC to manufacture your board. They say that engeneers will take a look at your data before manufactureig anyway ...
And to be honst, their preview sucks, in my eyes not out of alpha stage at this point ...

TT-392, 2022-05-31:
Ya, I know, I am just a bit paranoid I guess. Also, it acts as a bit of a sanity check as well after all. You can easily see if you chose the part number with the wrong footprint and stuff like that. But ye, more of a nice to have than a necessity.

computergeek1507, 2022-06-08:
It's probably more a JLC issue, from December - March-ish it lined up perfect, now it's been all over the place.

Bouni, 2022-06-08:
I guess they will completely refactor it, I've seen a screenshot from the chinese version of JLC somewhere and it looks way better and has a ton more options. So for now I'll use my time elsewhere as this is not a "real" problem ...

Bouni, 2022-12-30:
I think this can be closed as it works for me and I've not heared of any peroblems so far ...

closed 2022-12-30 by Bouni