Bouni/kicad-jlcpcb-tools #163: KiCad process does not exit when launching jlcpcb plugin

bvernoux, 2022-05-03:
**Describe the bug**
I'm using official KiCad 6.0.4 64bits on Windows 10
When I launch KiCad, load a board(PCB editor (pcbnew)) then I click on JLCPCB plugin(latest dev version [c528b61](https://github.com/Bouni/kicad-jlcpcb-tools/commit/c528b61e130cdc3b4073871c776869699a564ffb)) I close it immediately, then I close pcbnew & I close KiCad main window, I see (with Process Explorer v16.43) KiCad instance is not terminated and is still running, so it is a big issue as If I start an other KiCad instance I have 2 running instance in parallel and so on every time I just call/use the JLCPCB plugin.

For information I do not have any visible issue with JLCPCB plugin which perfectly it just seems to do not cleanup some things when we close it.

If I do not use JLCPCB plugin the KiCad instance is correctly terminated when I close KiCad.

Note: I have tested with other Python Plugin like "Interactive Html Bom" and I do not have that issue

**To Reproduce**
Steps to reproduce the behavior:
1. Launch KiCad 6.0.4 from icon
2. Load a project and click on the board "*.kicad._pcb" to open the PCB editor (pcbnew) window
3. Launch JLCPCB plugin then close it
4. Close pcb_new (Discard change if any) and close KiCad main window
5. Wait few seconds (in theory KiCad is fully closed)
6. Check with Process Explorer (or default process explorer of windows 10) and the kicad process is still present and running

**Expected behavior**
KiCad process shall exit properly when we close KiCad

# KiCad Version
```
Application: KiCad (64-bit)

Version: (6.0.4), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Mar 18 2022 03:04:44
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.6.0
	Curl: 7.78.0-DEV
	ngspice: 36
	Compiler: Visual C++ 1928 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Bouni, 2022-05-04:
I have seen this behaviour myself but not in a consistent way so that I can somehow reproduce it ...
I can confirm that it happens in 6.0.4 as well as in the 6.99 nightly versions.

bvernoux, 2022-05-04:
I confirm it is fixed with https://github.com/Bouni/kicad-jlcpcb-tools/commit/33154c02ae62da9a6a23df46e7ecde99e936a705

PR #164 (2022-05-04): Destroy dialog
This should fix #163
commit 2e2cb0ae (2022-05-04): Clean destroy Dialog on close event to fix #163
closed 2022-05-04 by Bouni