Bouni/kicad-jlcpcb-tools #165: Part Details dialog can't get input focus

alibenpeng, 2022-05-05:
**Describe the bug**
The "Part Details" dialog opens but never gets the input focus until the parent ("Select Part"/"JLCPCB Library", in my case) window is closed.

**To Reproduce**
Steps to reproduce the behavior:
1. Open JLCPCB Tools via toolbar button
2. Select a part from the list
3. Click on "Select Part"
4. Find and select a part in "JLCPCB Library" window
5. Click on "Part Details"
6. The window opens but doesn't get focus, and can't even be closed again as long as the parent window is open.

**Expected behavior**
The "Part Details" dialog should behave like any regular window.

# KiCad Version

```
Version: 6.0.4-6f826c9f35~116~ubuntu20.04.1, release build
```

livigni, 2022-05-08:
same issue here

Bouni, 2022-05-09:
I'll take a look at this as soon as possible ..

ayourk, 2022-05-17:
As an aside to this, only when I close the "JLCPCB Library" window can I then click on the "Open Datasheet" button in the "JLPCB Part Details" window.  Also, While the "JLCPCB Library" window is open, I cannot close any "JLPCB Part Details" windows until the "JLCPCB Library" window is closed.  Maybe the parent window is becoming modal or something like that?

Bouni, 2022-05-17:
@ayourk 🤔 maybe thats part off this issue. I need to look into this but first have to setup a linux machine including desktop which takes some time. Maybe I invest some more time and create a vm that I cna use for testing in the future ...

Bouni, 2022-06-07:
I finally can reproduce it in a Ubuntu VM. The strange thig is that if the part details windo is opend from the main window, everything works as expected, but if I start it from the select part window the behaviour is as you've described it ....

Bouni, 2022-06-07:
I tried now various things but its still a mystery to me whats going on here 😞 
Maybe I'm able to figure it out in the log run but at the moment I have not the slightes idea what else to try ....

livigni, 2022-06-07:
I have to say that I get same behaviour on macOs, so it shouldn't be a linux only thing. 
The thing is happening when I'm searching for a component and then I select one option and click on part details. Then it is not possible to get focus on that window, as it appears below the Search one. 
I believe that somehow the window got opened thinking it was called by the main Table view...

 
I explained myself terribly so I hope you could get some of what I meant :) Unfortunately I cannot send some screencast now as I am without external display and the plugin icon cannot be clicked (I think I will open another bug on this).

ayourk, 2022-06-07:
MacOS is a Unix based OS these days, so not sure how "Windows" related it is.

Bouni, 2022-06-08:
This is solved in the latest release.

closed 2022-06-08 by Bouni