Bouni/kicad-jlcpcb-tools #166: AttributeError: partially initialized module 'wx' has no attribute '__version__' (most likely due to a circular import)

busti, 2022-05-13:
**Describe the bug**
The Plugin fails to load on my system (I use nixos btw)

**To Reproduce**
Steps to reproduce the behavior:
1. Install KiCad on a fresh NixOS install using home-manager.
2. Install the plugin as explained in the installation instructions.
3. Open the PBC Editor.
4. The button should not be present.
5. Open Preferences / Action Plugins / Show Plugin Errors
6. The following error messages appear

```
  File "/nix/store/01kia41csjia67pry1rv828i9pvnnqfq-python3-3.9.12/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/nix/store/2gpnrvl6ayi6vsd06ijl4mxfrizfwhw4-kicad-base-6.0.5/share/kicad/scripting/kicad_pyshell/__init__.py", line 20, in <module>
    import wx
  File "/nix/store/whn1pfxdv1b5vmfyqbv37l4zbvvq4w01-python3.9-wxPython-4.1.1/lib/python3.9/site-packages/wx/__init__.py", line 12, in <module>
    __version__ = wx.__version__.VERSION_STRING
AttributeError: partially initialized module 'wx' has no attribute '__version__' (most likely due to a circular import)

**********************************
  File "/nix/store/01kia41csjia67pry1rv828i9pvnnqfq-python3-3.9.12/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/mbust/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/__init__.py", line 4, in <module>
    from .plugin import JLCPCBPlugin
  File "/home/mbust/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 5, in <module>
    from .mainwindow import JLCPCBTools
  File "/home/mbust/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 6, in <module>
    import wx
  File "/nix/store/whn1pfxdv1b5vmfyqbv37l4zbvvq4w01-python3.9-wxPython-4.1.1/lib/python3.9/site-packages/wx/__init__.py", line 12, in <module>
    __version__ = wx.__version__.VERSION_STRING
AttributeError: partially initialized module 'wx' has no attribute '__version__' (most likely due to a circular import)
```

**Expected behavior**
The plugin initializes successfully

# KiCad Version

```
Application: KiCad PCB Editor

Version: 6.0.5, release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.83.0 OpenSSL/1.1.1o zlib/1.2.12 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.2 libssh2/1.10.0 nghttp2/1.47.0

Platform: Linux 5.13.4 x86_64, 64 bit, Little endian, wxGTK, gnome, wayland

Build Info:
	Date: Jan  1 1980 00:00:00
	wxWidgets: 3.1.5 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.77.0
	OCC: 7.5.1
	Curl: 7.83.0
	ngspice: 36
	Compiler: GCC 11.3.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

Bouni, 2022-05-13:
I think I had a simmilar error in the past but cannot find the related issue right now.
As far as I remember it was something with the packaged wxpython version.
I try to reproduce it but I don't have a NixOS at hand (despite that's on my need-to-take-a-look-at list for quite a while 😏 )
Maybe it happens on a Ubuntu or Arch installation as well ...

busti, 2022-05-13:
Thanks a lot for the quick reply.
Can you tell me which version of wxpython is required for the plugin?  
NixOS makes it relatively easy to switch that out.

Bouni, 2022-05-17:
Hey, sorry for the delay. Actually I'm not sure and need some more time to investigate whats the casue of this ...

Bouni, 2022-06-07:
@busti I finally managed to set up a VM 😅 
If I install 6.0.5 from the Ubuntu PPA, then import wx via the KiCAD python shell and run `wx.__version__` I get `4.0.7` as result.
So if you get NixOS to use 4.0.7 instead of 4.1.1 you might get this to work ...

evils, 2022-08-16:
just a note since i found this after searching for the error i'm getting on opening the python terminal in the pcb editor

this is a long-standing issue with kicad using the latest wxwidgets and wxpython in nixpkgs
(very likely nothing to do with any plugin)
https://github.com/NixOS/nixpkgs/issues/90654

at least at first sight, this persists even when using wxwidgets 3.2.0 and wxpyhon 4.2.0 (latest release of both)

it may however be possible to use plugins and the python terminal with wxwidgets 3.0

closed 2024-10-11 by Bouni