Bouni/kicad-jlcpcb-tools #167: Untented vias not supported/honored

Engineer8468, 2022-05-16:
In my "Fabrication Outputs" dialog I have "Do not tent vias" checked, but the gerber outputs still tent the vias.  Can this plugin be changed to respect this setting?

Bouni, 2022-05-17:
Sure, can you test if the the [PR #168](#168 ) solves your problem?

Bouni, 2022-07-19:
@livigni Sorry for the delay. I think I need to add some sort of config option to allow users to select if they want tented vias or not.
AS a quick workaround you can edit this line https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/fabrication.py#L114 from `True` to `False`

ColinBathe, 2022-07-28:
Can I add a vote to make this optional please!

It would also be useful to know where the fabrication.py file ends up on windows? I can't find it!

Thanks. Love the plugin.

Bouni, 2022-07-28:
Just to keep you up to date, this is WIP!
I started changing all the buttons for toolbars as the make my life much easier. However that's a lot of work and takes some time.

Here's a quck peek:

[image]

And the settings menu with experimental toggle switches:

[image]

Not sure if I go further down that path or simply use checkboxes 🤷🏽

Oliver0804, 2022-07-29:
Hi sir,
If I change this line to false, can I change the via to have the ink cover the holes?
      
 popt.SetPlotViaOnMaskLayer(True) # Do not tent vias

I look forward to you adding this feature to the export options..<3

Bouni, 2022-07-29:
> Hi sir,
> If I change this line to false, can I change the via to have the ink cover the holes?
>       
>  popt.SetPlotViaOnMaskLayer(True) # Do not tent vias
> 
> I look forward to you adding this feature to the export options..<3
> 

Yes, exactly!

riskable, 2022-08-01:
Just want to confirm that my gerber files generated via JLCPCB tools are generating without tented vias which doesn't match my plotting settings.  It seems that no matter what my settings are the tools still get generated with "Do not tent vias" enabled (the opposite of what I want).

Bouni, 2022-08-08:
@riskable Your KiCad settings have nothing to do with what the plugin does! The plugin connects via the underlaying Python API and all settings have to be choosen by the plugin itself and to my knowledge there's no way to read the KiCAD settings.

arturo182, 2022-08-15:
This explains why all my boards suddenly have untested vias :D

nullstalgia, 2022-08-22:
Was a bit of a surprise for me too, but not an unpleasant one as it actually helped with some debugging. Love the new UI!

Bouni, 2022-08-23:
I'll set `popt.SetPlotViaOnMaskLayer` to false again by default and release a new version. As far as I remember, there's only one user that requested untented vias (and to be honest, I didn't thought about it very much when I made that change) but the rest seems to prefer tented vias.

Those wo wnat them untented can edit the soucre by themselves to set it from `False` to `True` if the really need it until the new UI is done.

arturo182, 2022-08-23:
With great plugin comes great responsibility :D

Bouni, 2022-08-29:
@arturo182 Indeed, I never expected this to be so popular and my spare time is way sparse compared to when I've started this 🙈

vargavik, 2022-10-10:
> Can I add a vote to make this optional please!
> 
> It would also be useful to know where the fabrication.py file ends up on windows? I can't find it!
> 
> Thanks. Love the plugin.

On Windows it is under <User Documents Folder>\KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools

ColinBathe, 2022-10-10:
> > Can I add a vote to make this optional please!
> > It would also be useful to know where the fabrication.py file ends up on windows? I can't find it!
> > Thanks. Love the plugin.
> 
> On Windows it is under \KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools

Not on my machine it isn't.

I don't have the 3rdparty directory.

vargavik, 2022-10-10:
> Not on my machine it isn't.
>
> I don't have the 3rdparty directory.

Ohh, for some reason the first part of the path is missing. You can find the Kicad Directory under your User Documents folder

Bouni, 2022-10-24:
You can use `%userprofile%` in the path so that you have to type a little bit less to get to the right folder lke this:

`%userprofile%\Documents\KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools`

Bouni, 2023-05-26:
This can be chosen via the settings menu. I'll close this now.

closed 2023-05-26 by Bouni