Bouni/kicad-jlcpcb-tools #176: No newer versions visible in KiCad Plugin and Content Manager

DomoticX, 2022-06-11:
Is it possible to add the newer versions/reases in the KiCad Plugin and Content Manager?

It saves time to manually upgrade ;-)

At the moment the older 2022.02.04 version in the latest in the list...

Kicad 6.0.5 / Win10 64-bit

[image]

livigni, 2022-06-11:
if you go in the "Repository" tab and you select the Bouni's repository,
this will fetch new updates (so the list will show new versions too).
Still, unfortunately the plugin will not update somehow: you still need to
uninstall, apply changes and install the new version. Maybe because it is
"testing" version and not "stable"?

ayourk, 2022-06-13:
Kicad proper has plans on making this better in the future: https://gitlab.com/kicad/code/kicad/-/issues/11552

Bouni, 2022-06-13:
@livigni and @ayourk nailed it, the only way to update is to first update the versions using the repository tab and then uninstall the old version and install the new version. Hopefully the KiCAD developers come up with a nice way of handling this soon.
Unfortunately there's nothing I can do about this for now, sorry 🤷🏽‍♂️

closed 2022-06-13 by Bouni