Bouni/kicad-jlcpcb-tools #178: Auto Refresh button or anytime Plugin is launched?

bcarriveau, 2022-06-20:
For example if the plugin has been run on a project it stores the part numbers it pulled in from the properties. But now if you make changes to the project the next time you run the plugin it does not pull in the new part numbers unless you remove the part, close and relaunch the JLCpcb plugin.

I now find myself just removing all the parts and relaunching to let it repopulate but at times doing this the plugin closes/crashes. 

Maybe this is already there and I'm missing it.?

Thanks!

Bouni, 2022-06-20:
Let me try to explain why I did things as they are done: 

A user creats the schematic, adds an LCSC field to all the parts and populates it by hand with the LCSC numbers. Then goes on to the PCB design, all the LCSC properties are transfered to the footprints when the update from schmematic functionality is used. When the plugin is started, all the footprints are checked for LCSC numbers and that info is then stored in the project database. From here you still have the option to select a different LCSC part as originally selected in the schematic.
On update we do not overwrite the info in the project database from the footprint because that would make the selection from the plugin useless. 

At the moment there is no way to write the selection back to the schematic, that would imediately solve this issue.

The KiCAD team said that they will develop such functionality in the future but not said when this will be done.
Hopefully in 6.99 aka 7.0

Maybe I should add a additional field where I store the source of the LCSC ID, lets call it `schematic` when the number came from the schematic. When a user selects another part using the plugin, that field is changed to `plugin`. On Update, all fields with `schematic` are updated from the schematic, the ones with `plugin` are left untouched.

Would such a feature fix the issue for you?

Bouni, 2022-06-21:
If the KiCAD devs enable me to write data back to the schematic in the future I'll defenitely do that to improve the usability, but right now I'm kind of limited 🤷🏽 
Anyway, glad you can work with it for now 😅

closed 2022-06-21 by Bouni