Bouni/kicad-jlcpcb-tools #179: Issue with latest KiCad 6.0.6 Windows10 64bits

bvernoux, 2022-06-20:
**Describe the bug**
When lauching kicad-jlcpcb-tools an error windows appears
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'PCB Editor'
2. Click on 'kicad-jlcpcb-tools' icon

4. See error
[image]

**Expected behavior**
The plugin shall work like on previous version KiCad 6.0.5

# KiCad Version
```
Application: KiCad PCB Editor (64-bit)

Version: (6.0.6), release build

Libraries:
	wxWidgets 3.1.6
	libcurl/7.82.0-DEV Schannel zlib/1.2.12

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jun 18 2022 23:26:20
	wxWidgets: 3.1.6 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.0
	Curl: 7.82.0-DEV
	ngspice: 37
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Bouni, 2022-06-20:
@bvernoux Please test the referenced PR, that should fix your issue.

bvernoux, 2022-06-21:
I confirm https://github.com/Bouni/kicad-jlcpcb-tools/pull/180 solve the issue on KiCad 6.0.6

Bouni, 2022-06-21:
Nice, thanks for your feedback!

closed 2022-06-21 by Bouni