Bouni/kicad-jlcpcb-tools #181: Error:  AttributeError: module 'wx' has no attribute 'BitmapBundle'

timcooijmans, 2022-06-21:
Hi, 

Thank you for this excellent tool. With the latest version (6.0.3) I get the following error while trying to use the plugin. Sadly I never used it before on OS X, so not sure if it's a new error or not. It seems to work on Windows.

```
Traceback (most recent call last):

File "/Users/username/Documents/KiCad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 21, in Run
dialog = JLCPCBTools(None)

File "/Users/username/Documents/KiCad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 183, in __init__
loadBitmapScaled(

File "/Users/username/Documents/KiCad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/helpers.py", line 53, in loadBitmapScaled
return wx.BitmapBundle(bmp)

AttributeError: module 'wx' has no attribute 'BitmapBundle'
```

ayourk, 2022-06-22:
I'm getting something similar: 
[image]

This happens with the latest 2022.06.03 release.  Reverting to the 2022.06.02 release fixes the issue (work around).

Bouni, 2022-06-22:
@timcooijmans @ayourk 

Please always include your KiCAD version Info in issues, without that Its almost impossible to figure out what happened!

KiCAD -> Help -> About Kicad -> Copy version Info

timcooijmans, 2022-06-22:
Hi, sorry:

```Application: KiCad

Version: (6.0.6-0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.79.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.45.1

Platform: macOS Version 12.4 (Build 21F79), 64 bit, Little endian, wxMac

Build Info:
	Date: Jun 19 2022 14:26:31
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.78.0
	OCC: 7.6.2
	Curl: 7.64.1
	ngspice: 35
	Compiler: Clang 12.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

ayourk, 2022-06-22:
And in case you need mine too:
```
Application: KiCad

Version: 6.0.5-a6ca702e91~116~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.12

Platform: Linux 5.15.0-37-generic x86_64, 64 bit, Little endian, wxGTK, , x11

Build Info:
	Date: May  3 2022 08:42:29
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.83.0
	ngspice: 36
	Compiler: GCC 11.2.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Bouni, 2022-06-22:
😭 It seems that all OSes ship different wx /wxwidgets versions which leads to this kind of errors. 
I thought that simply checking the KiCAD version is enough but I need to come up with a better plan 😕 
Maybe I check if wx has BitmapBundle and if so use that 🤔 

Give me a few hours to work something out!

kalanzun, 2022-06-23:
Hi! Same error for me on Arch Linux. wxWidgets documentation says, wxBitmapBundle is a new class in version 3.1.6

[https://docs.wxwidgets.org/trunk/classwx_bitmap_bundle.html](https://docs.wxwidgets.org/trunk/classwx_bitmap_bundle.html)

```
Application: KiCad Leiterplatteneditor

Version: (6.0.6), release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.83.1 OpenSSL/1.1.1o zlib/1.2.12 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.2 libpsl/0.21.1 (+libidn2/2.3.0) libssh2/1.10.0 nghttp2/1.47.0

Platform: Linux 5.18.5-zen1-1-zen x86_64, 64 bit, Little endian, wxGTK, gnome, wayland

Build Info:
	Date: Jun 21 2022 02:56:58
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.79.0
	OCC: 7.5.3
	Curl: 7.83.1
	ngspice: 37
	Compiler: GCC 12.1.0 with C++ ABI 1017

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Bouni, 2022-06-24:
@timcooijmans @ayourk @kalanzun As you three are on different platforms it would be awesome if you could give PR #184 a try and see if that fixes this issue.

livigni, 2022-06-25:
Same on Mac OS.

```
Traceback (most recent call last):
File "/Users/vicio/Documents/KiCad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 21, in Run
dialog = JLCPCBTools(None)

File "/Users/vicio/Documents/KiCad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 183, in __init__
loadBitmapScaled(

File "/Users/vicio/Documents/KiCad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/helpers.py", line 53, in loadBitmapScaled
return wx.BitmapBundle(bmp)

AttributeError: module 'wx' has no attribute 'BitmapBundle'
```
```
Application: KiCad PCB Editor

Version: (6.0.6-1)-1, release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.77.0 (SecureTransport) LibreSSL/2.8.3 zlib/1.2.11 nghttp2/1.42.0

Platform: macOS Version 12.1 (Build 21C52), 64 bit, Little endian, wxMac

Build Info:
	Date: Jun 20 2022 17:18:13
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.78.0
	OCC: 7.6.2
	Curl: 7.64.1
	ngspice: 35
	Compiler: Clang 12.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

ayourk, 2022-06-26:
> @timcooijmans @ayourk @kalanzun As you three are on different platforms it would be awesome if you could give PR #184 a try and see if that fixes this issue.

What is the build process so I can add this via a Zip file to Kicad?

Bouni, 2022-06-26:
You have to install the plugin using git and not via PCM.
Then you can fetch the PR branch and switch to it.

ayourk, 2022-06-26:
> You have to install the plugin using git and not via PCM. Then you can fetch the PR branch and switch to it.

I downloaded the branch .zip file.  How should I reconstruct the .zip?

kalanzun, 2022-06-27:
You may just unpack the zip to your local plugin directory. `~/.local/share/kicad/6.0/scripting/plugins/`. KiCAD can run the project directly.

PR #184 (2022-06-29): Check wxWidgets version 
Check wxWidgets version in order to decide wether to use Bitmap or BitmapBundle for button icons and static Icons.
Hopefully fixes #181 on all platforms 🤞🏽
closed 2022-06-29 by Bouni