Bouni/kicad-jlcpcb-tools #189: assertion error when opening the plugin from kicad 6.07

lamikr, 2022-06-30:
When I open the plugin, I get assertion error:

./src/common/sizer.cpp(2118): assert "!(flags & wxALIGN_RIGHT)" failed in DoInsert(): Horizontal alignment flags are ignored in horizontal sizers

This is the full backtrace:

ASSERT INFO:
./src/common/sizer.cpp(2118): assert "!(flags & wxALIGN_RIGHT)" failed in DoInsert(): Horizontal alignment flags are ignored in horizontal sizers

BACKTRACE:
[1] wxBoxSizer::DoInsert(unsigned long, wxSizerItem*)
[2] _PyObject_MakeTpCall
[3] _PyEval_EvalFrameDefault
[4] _PyObject_FastCallDict
[5] _PyObject_MakeTpCall
[6] _PyEval_EvalFrameDefault
[7] _PyObject_FastCallDict
[8] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[9] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[10] wxEvtHandler::TryHereOnly(wxEvent&)
[11] wxEvtHandler::DoTryChain(wxEvent&)
[12] wxEvtHandler::ProcessEvent(wxEvent&)
[13] wxWindowBase::TryAfter(wxEvent&)
[14] wxAuiToolBar::OnLeftUp(wxMouseEvent&)
[15] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[16] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[17] wxEvtHandler::TryHereOnly(wxEvent&)
[18] wxEvtHandler::ProcessEventLocally(wxEvent&)
[19] wxEvtHandler::ProcessEvent(wxEvent&)
[20] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[21] g_closure_invoke
[22] g_signal_emit_valist
[23] g_signal_emit
[24] gtk_main_do_event
[25] g_main_context_dispatch
[26] g_main_loop_run
[27] gtk_main
[28] wxGUIEventLoop::DoRun()
[29] wxEventLoopBase::Run()
[30] wxAppConsoleBase::MainLoop()
[31] APP_KICAD::OnRun() /usr/src/debug/kicad-6.0.07-1.mga8.x86_64/kicad/kicad.cpp:317
[32] wxEntry(int&, wchar_t**)
[33] main /usr/src/debug/kicad-6.0.07-1.mga8.x86_64/kicad/kicad.cpp:398
[34] __libc_start_main
[35] _start

**To Reproduce**
Steps to reproduce the behavior:
1. Open PCB
2. Open Plugin
3. See error

**Expected behavior**
To be open the plugin without pressing the Continue button after assert error

# KiCad Version

Application: KiCad PCB Editor

Version: 6.0.7-1-dev-87d4249d9f, debug build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.74.0 OpenSSL/1.1.1o zlib/1.2.12 brotli/1.0.9 libidn2/2.3.0 libpsl/0.21.1 (+libidn2/2.3.0) libssh/0.9.6/gnutls/zlib nghttp2/1.42.0

Platform: Linux 5.15.46-desktop-1.mga8 x86_64, 64 bit, Little endian, wxGTK, gnome, wayland

Build Info:
	Date: Jun 26 2022 01:45:27
	wxWidgets: 3.1.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.75.0
	OCC: 7.4.1
	Curl: 7.74.0
	ngspice: 33
	Compiler: GCC 10.3.0 with C++ ABI 1014

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
	KICAD_STDLIB_DEBUG=OFF
	KICAD_STDLIB_LIGHT_DEBUG=OFF
	KICAD_SANITIZE_ADDRESS=OFF
	KICAD_SANITIZE_THREADS=OFF

wxPython version is 4.1.1 (python3-wxpython4-4.1.1-1.1.mga8)
python version is 3.8.12 
Distribution: Mageia 8 Linux 
Plugin: latest version of jlcpcb plugin from github.

lamikr, 2022-06-30:
When I open the plugin I see the error only once per opening. But when I open the
"select part" dialog I see multiple time a similar type of error:

./src/common/sizer.cpp(2097): assert "!(flags & wxALIGN_CENTRE_VERTICAL)" failed in DoInsert(): Vertical alignment flags are ignored in vertical sizers

Bouni, 2022-07-01:
Its strange that you see C++ errors but thats maybe from the bindings used in that particular version.
I have a feeling for whats going on here.
If you look at these two lines: https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/partselector.py#L220-L221
you see that there's a vertical sizer and the first element added to it gets a ALIGN_CENTRE_VERTICAL flag set.
That has not raised any erros so far.
I could potentially remove them but need to make sure that this does not cripple the layout for everybody else ...

Is there a reason why you run the dev version of 6.0?

Bouni, 2024-10-11:
I'll close this. KiCAD 6 is no longer supported 🤷🏽‍♂️

closed 2024-10-11 by Bouni