Bouni/kicad-jlcpcb-tools #193: Please add "LCPC Part webpage" button.

ayourk, 2022-07-06:
**Is your feature request related to a problem? Please describe.**
Please add a button to go to the LCPC web page based upon LCPC part number in the same places where the Datasheet button is also placed.

ayourk, 2022-07-06:
Only thing I could see as a potential problem is how to open the "default" browser window in all the different platforms.

Bouni, 2022-07-06:
Thats not a big deal, we'll do it the same way we open the datasheet: https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/partdetails.py#L121

DanielO, 2023-03-18:
I had look at this but it's harder than it should be because the 'lcscGoodsUrl' key does not give you a working URL (unlike the datasheet one). eg for 'C60133' you get:
`"https://www.lcsc.com/product-detail/multilayer ceramic capacitors mlcc - smd/smt_yageo_cc0402krx7r9bb103_C60133.html/?href=jlc-SMT/?href=jlc-SMT"`

but it should be:
`https://www.lcsc.com/product-detail/Multilayer-Ceramic-Capacitors-MLCC-SMD-SMT_YAGEO-CC0402KRX7R9BB103_C60133.html`

I had a go at decoding the JS called when you search for something but didn't have much luck.

Bouni, 2024-10-11:
I just tested this and for me the `lcscGoodsUrl` key does work!?