Bouni/kicad-jlcpcb-tools #197: Manually fill in a LCSC part number

SwooshMihu, 2022-07-25:
Certain components show up in the online LCSC shop but not in the embedded tool. 
Is there any way to manually fill in the LCSC part number?
[image]

SwooshMihu, 2022-07-25:
Apparently i can right click on the LCSC cell and select "Paste LCSC".

FDSoftware, 2022-07-25:
the stock in LCSC and JLCPCB differ a bit, this tool uses the JLCPCB stock, you can check it from here:

https://jlcpcb.com/parts
https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=C2925591

Bouni, 2022-07-26:
To my knowledge you can't use all LCSC parts on your designs to be assembled, just the subset that is listed by JLC.
@SwooshMihu / @FDSoftware Have you used non JLC listed LCSC parts in your designs before and they assembled them for you?

SwooshMihu, 2022-07-26:
From what i know, they will use any part from LCSC in the assembly process if you can find it in stock and provide the link for it.

FDSoftware, 2022-08-05:
@Bouni
only the subset can be used, but there are several cases where you can send them an email to check if it is pending restock (part in the JLCPCB library but not in stock), but it is not something "sure" that it will always happen

https://support.jlcpcb.com/article/99-does-the-red-dot-means-pin-1-in-the-placement-previewer

> 6. Why there is in stock on LCSC website but not in JLCPCB library?
The reason is that our components sources are from LCSC, and LCSC has two warehouses located in Jiangsu and Shenzhen. So sometimes the components are shown available on LCSC website, but it shows 0 when you are placing an order. This situation is caused by different warehouses sources, and we have no way to transfer them.

z2amiller, 2026-09-26:
On the open question above: JLC can assemble LCSC parts that are not in its SMT library. Through Parts Pre-order / Global Sourcing it buys them from LCSC for the order, and they are then placed like any other part. The plugin's catalog only lists JLC's library, so those numbers never show up in the part selector.

#871 adds a way to assign one: right-click > **Enter LCSC…** takes a number or a pasted lcsc.com / jlcpcb.com link, and the part selector offers an assign-anyway button when a searched number isn't among its results.

closed 2026-09-28 by chmorgan