Bouni/kicad-jlcpcb-tools #201: X,Y of THT components are off

DomoticX, 2022-08-03:
I know i saw this issue before here somewhere, but seems closed?

So, i added some THT components like JST, RJ45 and BARREL connectors, when i export the files and upload them to JLCPCB, the POS X,Y of the THT components are off.

I remember something about some components have a different "pin1" start position, but is this even fixable?, that would be great!

[image]

---------------------------------------------------
Application: KiCad PCB Editor (64-bit)

Version: (6.0.7), release build

Libraries:
	wxWidgets 3.1.7
	libcurl/7.83.1-DEV Schannel zlib/1.2.12

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jul 26 2022 02:49:38
	wxWidgets: 3.1.7 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.0
	Curl: 7.83.1-DEV
	ngspice: 37
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

DomoticX, 2022-08-03:
I was thinking today about a solution for this..., what if we have like a "X,Y offset" list, just like the existing "rotation" list, in here you could place the extra offset (in MM) for a item name (regex) or item code from LCSC.

Bouni, 2022-08-08:
The main issue is that JLC expects the coordinates to be in the center of the model where KiCAD parts are most of the time but not always have their center on Pin 1.

I think it would be possible to fix that automaticaly by calculate the component center and then add an autmoatic offste but thats a lot of work. 

Your suggestion could be easy implemented, so I guess I'll go for that ....

DomoticX, 2022-08-08:
Great!, thank you!, a tip: please add the X,Y correction after the "rotation" correction is done ;-)

DomoticX, 2022-09-30:
Ps. today i saw that KIKIT also implemented such feature, maybe you can use a bit of that code (would save some time) ;-)

https://blog.honzamrazek.cz/2022/09/kikit-v1-1-whats-new/

[image]

computergeek1507, 2023-01-29:
This was mostly fixed when I switched it using the center of the bounding box for Thur hole parts during exports, parts over hanging the edge still had issues. 217580d2045772c84c3322a97af129dcb226a4f4

chmorgan, 2025-04-01:
Closing due to inactivity.

closed 2025-04-01 by chmorgan