Bouni/kicad-jlcpcb-tools #209: Public to Kicad Plugin and Content Manager

computergeek1507, 2022-08-10:
Rearrange the github structure to match the kicad docs so the plugin can be added to the Content Manager

https://dev-docs.kicad.org/en/addons/

Bouni, 2022-08-10:
Will not happen as the Kicad Team considers this a commercial plugin which needs approval from JLCPCB.

Check out the commercial services section of this: https://dev-docs.kicad.org/en/addons/

JLC made it clear to me that the will not do that as they focus on dragging people into easyeda.

computergeek1507, 2022-08-10:
Benny Megidish's JLC-Plugin-for-KiCad is currently in there

Bouni, 2022-08-12:
This is because Benny Megidish's plugin does not directly interact with JLC PCB APIs, just formats the output files in the correct way. As this plugin does use JLC API's the Addon team will not merge it into the official PCM repo unless JLC gives them permission which they will not 😞 

See https://gitlab.com/kicad/addons/metadata/-/merge_requests/93

chmorgan, 2025-04-01:
Closing due to inactivity.

closed 2025-04-01 by chmorgan