Bouni/kicad-jlcpcb-tools #211: Cannot Update Plugin

computergeek1507, 2022-08-23:
**Describe the bug**
I don't know if this is a Kicad bug or an issue with the repo or I don't know what I am doing. I cannot update the plugin through the content manager. If I select the row with the newest version the install button is still greyed out. The only way I found to "upgrade" is uninstall the plugin and reinstall it. This deletes the database, so I manually back it up before I do this.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

[image]

**Expected behavior**
Install button is not greyed out when selecting another version.

# KiCad Version

```
Application: KiCad (64-bit)

Version: (6.0.7), release build

Libraries:
	wxWidgets 3.1.7
	libcurl/7.83.1-DEV Schannel zlib/1.2.12

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jul 26 2022 02:49:38
	wxWidgets: 3.1.7 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.0
	Curl: 7.83.1-DEV
	ngspice: 37
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

Bouni, 2022-08-24:
That's strange, I've tested this on the same setup as you have and it worked forme on the first try:
[image]

Bouni, 2022-08-24:
If I remember correctly, you have to uninstall the old version before you're able to install a new one.
There's a open issue for that on Kicad's GitLab which I don't find at the moment-

Bouni, 2022-09-05:
@computergeek1507 Can you verify that updating is possible by removing and reinstalling?

MichaelRColton, 2022-11-16:
I had the same issue of not being able to update directly, but uninstalling and reinstalling works. That seemed to occur with other plugins as well, so yeah, maybe that's on KiCad's end.

chmorgan, 2025-04-01:
Closing due to inactivity.

closed 2025-04-01 by chmorgan