Bouni/kicad-jlcpcb-tools #217: JLCPCB 7-digit Manufacturer Part #'s Will Not Show in Plugin

alexquilty, 2022-09-05:
**Describe the bug**
When I search for parts from LCSC with Part #'s consisting of 7 digits after the "C", the part will not show despite the part being In Stock, Extended, or Basic.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'LCSC Plugin'
2. Click on 'Select Part' for any of the PCB components
3. Enter any LCSC component with any of the filters and make sure the components have 7 numbers following the 'C' for example:
C7138442
4. Observe the component not appearing despite being In-Stack, and Basic or Extended

**Expected behavior**
Component should appear in the listing.

# KiCad Version

```
Application: KiCad PCB Editor (64-bit)
Version: (6.0.7), release build
Libraries:
    wxWidgets 3.1.7
    libcurl/7.83.1-DEV Schannel zlib/1.2.12
Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Date: Jul 26 2022 02:49:38
    wxWidgets: 3.1.7 (wchar_t,wx containers)
    Boost: 1.79.0
    OCC: 7.6.0
    Curl: 7.83.1-DEV
    ngspice: 37
    Compiler: Visual C++ 1929 without C++ ABI
Build settings:
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON
```

alexquilty, 2022-09-05:
Wasn't sure if this was a bug and should be fixed, or if it would technically be a new feature request if it is not implemented yet. My guess was that it would be a bug technically since it is a feature it should already have. Please let me know if you can fix this. There is no time crunch, and if there's anything I can do to help I will try to as much as I can to help keep this amazing plugin going for KiCAD users.

Bouni, 2022-09-06:
Hi, I think this is a common misconception about the LCSC numbers as LCSC numbers != JLCPCB numbers.

The JLCPCB parts database is just a subset of the entire LCSC portfolio of parts, so not all of them are officially available for assembly by JLC.

[image]

Another user already requested that all LCSC numbers should be available in this plugin, but that brings a problem with it.
At the moment we search the parts db supplied by JLC, thats why your numer is not found. If we want to go for all LCSC numbers we need a new data source which we don't have at the moment.

Furthermore, how can we make sure that JLC will assemble a part if its a "LCSC only" part!?

Let me know what you think ...

Bouni, 2023-01-04:
I checked with JLC via their chat function:

```
**JLCPCB**
What do you need help with? PCB Assembly
Hello. How may I help you?

**me**	
Hi, can I include all LCSC parts in my design for assembly or just those listed by JLC in the parts section?
Specificly C5122489

**JLCPCB**
please note we support 1 side or 2 sided assembly now (if need 2 sides assembly, pls choose "standard assembly"). and that only parts in our part lib(jlcpcb.com/parts) and have stock will be selected, and only selected parts will be charged and assembled. And for the through hole parts, also need be in our part lib and have in stock, and make them in the Bom and CPl with same designator and similar coordinates, once it is selected, will be mounted as well.
so pls check if they are also in our part lib,. thanks

**me**	
Ok, thank you for clarification!

**JLCPCB**
you are welcome friend
and if no stock , you may try these 2 ways
for some parts without stock or not in our part lib, you may try these 2 features (but should meet the MOQ and attrition for assembly, for small parts, pls add 15 pcs if possible):
a) try part pre order as this link https://support.jlcpcb.com/article/151-how-to-build-your-own-parts-library-in-jlcpcb
b) Try global sourcing part purchasing https://support.jlcpcb.com/article/192-how-to-use-jlcpcb-global-sourcing-parts-service (first , the platforms such as Mouser, Digikey should have in stock, then we can order. and need the order of the component finished( in the part order history, it shown "in warehouse"), then you can submit the SMT order))
And option b , you may take it as priority, since other platforms are with stock, so we can buy from them directly
Pls note part pre-order or global sourcing part purchasing usually will be done in 9-15 working days usually.
And only the part purchasing is finished (part order history will show "complete") .Then pls submit the SMT order(put the part number in the bom and cpl, and the part will be selected from "my part",need to meet the MOQ and attrition) later. (more details pls refer to https://support.jlcpcb.com/article/212-how-to-use-my-own-parts-for-pcb-assembly-order)
if any other questions, just be free to ask , have a good day. bye 
```

If I get it right, only parts listed in their parts section are allowed. So I'll close this one.

closed 2023-01-04 by Bouni

alexquilty, 2023-01-04:
Got it, thanks for your help Bouni! I am a student over in university and I was working with an industrial advisor for my Electrical Engineering Senior Design PCB project, and I was just about ready to rip my hair out trying to find these parts lol. I eventually did find a replacement, and man your plugin for JLCPCB really saved my ass since I decided to use KiCAD haha. Anyway thank you again for all of your help and for making that plugin!

One thing I never did get down is that my imported gerber file into JLCPCB always had funky placements and wrong polarity for ICs, but it wasn’t too much of a serious issue since I just moved all the parts where they should be and double checked polarity w/ silkscreen I put on the board.

Sincerely,
Alexander Quilty