Bouni/kicad-jlcpcb-tools #219: Tinted via suddenly deactivated

glowingkitty, 2022-09-09:
I recently reset my computer and also reinstalled KiCad and the JLCPCB plugin. And suddenly the jlcpcb tools plugin exports the gerber files without tinted vias. Just double checked - when I export from KiCad with the default settings, the vias are tinted (which I also wanted to have).
Did you change the default export settings for your plugin? and how can one overwrite them?

Bouni, 2022-09-14:
What version do you run?
I recently set that back to tented by default which is inluded in the [latest version 2022.09.01](https://github.com/Bouni/kicad-jlcpcb-tools/blob/5c25592e47f8682dbe2c45d0d3fa5b8da4e3c253/fabrication.py#L114)

zaxour, 2022-09-23:
Tented vias are back! Thank you for fixing this!

Bouni, 2022-09-23:
So that means that @glowingkitty is using a older version then.
Please update to solve your issue!

closed 2022-09-23 by Bouni