Bouni/kicad-jlcpcb-tools #220: Generate fabrication file zip, includes the zip in itself

missikos, 2022-09-13:
**Describe the bug**
After pressing generate fabrication files, inspecting the output there is a zip, opening that zip file, there is another copy of itself in itself. I think that it shouldn't be there.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to tool
2. Click on Generate fabrication file, top left
3. Open the folder it created
4. Open the zip it created, see another zip inside the zip

**Expected behavior**
No duplication of zip

# KiCad Version
6.0.2-0

Bouni, 2022-09-19:
I had to revert my "fix" as it did break the zip file generation ...

To reproduce your issue I updated both Kicad 6.0 to the latest version (6.0.7) and 6.99 to the latest nightly build.
In neither of these I was able to find the zip containing itself.

Can you update to 6.07 as well an test again?

computergeek1507, 2022-10-13:
Maybe this is OS related, I used Kicad 6.0.8 and Windows 10 and do not see this issue

commit 84233065 (2022-09-15): Fix #220, prevent zip file from containing the zip itself
commit cb7a2990 (2022-09-19): Revert "Fix #220, prevent zip file from containing the zip itself"

This reverts commit 84233065352aeb74e82025baeaddb7999a5b9990.
closed 2023-03-24 by Bouni