Bouni/kicad-jlcpcb-tools #223: Does not start at all in Debian testing/unstable

bzed, 2022-09-16:
**Describe the bug**
The plugin fails to start with the error shown in the image: (can't copy&paste it, doesn't make a difference anywya)

[image]

My python wx knowledge is not existat, I was basically not able to debug this in a useful amount of time, sorry.

**To Reproduce**
Steps to reproduce the behavior:
1. install Debian testing or unstable
2. install kicad from Debian (6.0.7)
3. git clone the plugin
4. run kicad
5. try to start the plugin

**Expected behavior**
A working plugin.

# KiCad Version

```
Application: KiCad PCB Editor
Version: 6.0.7+dfsg-1+b1, release build
Libraries:
    wxWidgets 3.0.5
    libcurl/7.85.0 OpenSSL/3.0.5 zlib/1.2.11 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.3 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.10.0 nghttp2/1.49.0 librtmp/2.3
Platform: Linux 5.18.0-2-amd64 x86_64, 64 bit, Little endian, wxGTK, i3, x11
Build Info:
    Date: Aug 3 2022 09:25:50
    wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.74.0
    OCC: 7.6.3
    Curl: 7.84.0
    ngspice: 37
    Compiler: GCC 12.1.0 with C++ ABI 1017
Build settings:
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON
```

bzed, 2022-09-18:
I've tried

```
diff --git a/plugin.py b/plugin.py
index 8393fd2..415bdb8 100644
--- a/plugin.py
+++ b/plugin.py
@@ -1,4 +1,5 @@
 import os
+import wx
 
 from pcbnew import ActionPlugin
 
@@ -21,3 +22,10 @@ class JLCPCBPlugin(ActionPlugin):
         dialog = JLCPCBTools(None)
         dialog.Center()
         dialog.Show()
+
+
+if 'phoenix' in wx.PlatformInfo:
+    if not wx.GetApp():
+        wx_app = wx.App()
+    else:
+        wx_app = wx.GetApp()
```

which seems to fix the app issue, but the result are more errors - seems wx changed a bit.

Bouni, 2022-09-19:
Kicad and wx / wxpython is a nightmare, every os / distro seems to package a slightly different version which leads to a ton of different way stuff needs to be handled 😭

bzed, 2022-09-19:
I know, once there was a time when I helped to package wx python for Debian, its a nightmare.

chmorgan, 2025-03-29:
Closing due to inactivity.

closed 2025-03-29 by chmorgan