Bouni/kicad-jlcpcb-tools #225: Stale/invalid mappings prevent pulling LCSC assignments from the schematic

ajf58, 2022-09-19:
**Describe the bug**

Sometimes the tool doesn't pick up LCSC numbers which it has previously written to the schematic. The slightly crowded screenshot below shows the issue. C3 has a part number assigned as shown in the bottom half of the image. In the PCB layout, the tool shows the stock level for this part (C107110), but nothing in the "LCSC" column in the table.

[image]

I think this is due to a conflict between the "mapping" file, and the information in the schematic. Deleting all of the mappings (via the "manage mappings" window) has resolved the problem (I think).

**KiCad Version**

```
Application: KiCad PCB Editor (64-bit)

Version: (6.0.5), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.82.0-DEV Schannel zlib/1.2.12

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: May  3 2022 00:36:16
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.0
	Curl: 7.82.0-DEV
	ngspice: 36
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

Bouni, 2022-09-27:
Does close and reopen the plugin help it that case?

ajf58, 2022-09-29:
Closing and reopening didn't seem to fix things. The problem went away when I deleted all the mappings (via the "manage mappings" menu), but I'm not very confident on what the exact solution was that made the symptom go away, unfortunately.

Bouni, 2022-09-30:
So that meany you no longer have a project that has these issue, right?
I guess there's no wa to figure out what went wrong without being able to reproduce it....
I'll close the issue and if you get the same kind of problems again, you can re-open.

closed 2022-09-30 by Bouni