Bouni/kicad-jlcpcb-tools #226: Exception on python action plugin code - AttributeError: "NoneType" object has no attribute "group"

jlillest, 2022-09-20:
**Describe the bug**
When trying to run the plugin from the Tools->External Plugins->JLPCB Tools button, it gives the following error in a popup window:
[image]

I was unable to find this error in the issue tracker.

**To Reproduce**
Steps to reproduce the behavior:
1. Install KiCad 6 on Ubuntu 18.04
2. Open the PCB Editor of a project
3. Install JLPCB Plugin from Plugin and Content Manager
4. Click on Tools->External Plugins->JLPCB Tools
5. See error

**Expected behavior**
I would expect the tool to open as described in the README

# KiCad Version
Ubuntu 18.04.6/bionic
```
Application: KiCad PCB Editor

Version: 6.0.4-6f826c9f35~116~ubuntu18.04.1, release build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.58.0 OpenSSL/1.1.1 zlib/1.2.11 libidn2/2.3.0 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3

Platform: Linux 4.15.0-162-generic x86_64, 64 bit, Little endian, wxGTK, xubuntu, x11

Build Info:
	Date: Mar 21 2022 17:21:28
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.22
	Boost: 1.65.1
	OCC: 7.5.2
	Curl: 7.81.0
	ngspice: unknown
	Compiler: GCC 7.5.0 with C++ ABI 1011

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

Bouni, 2022-09-22:
This most likely failes du to a very old wxwidgets version (wxWidgets 3.0.4) which probably is part of your OS which also quite old (current Ubuntu is 22.04 I think). Also the KiCAD version is not the latest stable version (6.0.7 is the latest at the moment).
Maybe updateing your OS will update wxwidgets as well but I don't know for sure

Bouni, 2023-01-18:
Have you tried upgrading your OS in the meantime?

chmorgan, 2025-04-01:
Closing due to inactivity.

closed 2025-04-01 by chmorgan