Bouni/kicad-jlcpcb-tools #228: use JLCPN for JLCPCB Assembly Parts Library codes

easyw, 2022-10-11:
Would it be possible to fetch for LCSC part number also for a JLCPN (JLC Part Number) tag/field?
Referring to https://jlcpcb.com/parts
the parts are from "JLCPCB Assembly Parts Library" ... 
In schematics I use to add JLCPN Field Name instead of LCSC PN because JLCPN are the one used for PCB assembly

easyw, 2022-10-11:
btw have you a working minimal sample to test in which way the LCSC field is read? Does this field has to be at a specific sort position?

Bouni, 2022-10-31:
At the moment all properties are converted to lower case and then checked if they contain `lcsc`, the first that matches this search is used:

https://github.com/Bouni/kicad-jlcpcb-tools/blob/e6e0a25bf1cf4fccabce23488d5cf8020b008c60/helpers.py#L79-L84

We could expand that to take `jlc` into account as well. Maybe It should be validated if the number matches a regex like `C\d+` to make sure only valid LCSC/JLC numbers are used.

Bouni, 2022-11-25:
Done, will be part of the next release

closed 2022-11-25 by Bouni