Bouni/kicad-jlcpcb-tools #229: Cannot Install on MacPorts KiCAD

ahalekelly, 2022-10-13:
**Describe the bug**
I installed the plugin via the PCM, however there's no icon in the toolbar. All my other plugins work (all from the official repository). 

**To Reproduce**
Install KiCAD on an M1 Mac via MacPorts
Go to the PCM, add custom repository, install KiCAD JLCPCB Tools
Log:
```Downloading package url: 'https://github.com/Bouni/kicad-jlcpcb-tools/releases/download/2022.09.01/KiCAD-PCM-2022.09.01.zip'
Extracting package 'com.github.bouni.kicad-jlcpcb-tools'.
Extracted package 
Removing downloaded archive 'com.github.bouni.kicad-jlcpcb-tools_v2022.09.01.zip'.
Done.
```
PCM says that the plugin is installed
Open PCBNew, no JLC button in toolbar.

I also tried manually installing via git with no luck. Is there any way to check and make sure it's installed?

**Expected behavior**
JLC icon should appear in toolbar

# KiCad Version
MacPorts KiCAD (ARM Build)

```
Application: KiCad

Version: 6.0.7, release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.85.0 OpenSSL/3.0.5 zlib/1.2.12 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.3 libpsl/0.21.1 (+libidn2/2.3.2) nghttp2/1.50.0

Platform: macOS Version 12.6 (Build 21G115), 64 bit, Little endian, wxMac

Build Info:
	Date: Sep 22 2022 05:42:28
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.6.0
	Curl: 7.85.0
	ngspice: 36
	Compiler: Clang 13.0.0 with C++ ABI 1002

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

Bouni, 2022-10-14:
Check preferences -> preferences -> action plugins

If there's a problem you should see a attention icon that gives more info.

Let me know if that's the case

ahalekelly, 2022-10-19:
Ah thank you!
```
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/Users/akelly/Documents/kicad/6.0/scripting/plugins/kicad-jlcpcb-tools-main/__init__.py", line 4, in <module>
    from .plugin import JLCPCBPlugin
  File "/Users/akelly/Documents/kicad/6.0/scripting/plugins/kicad-jlcpcb-tools-main/plugin.py", line 5, in <module>
    from .mainwindow import JLCPCBTools
  File "/Users/akelly/Documents/kicad/6.0/scripting/plugins/kicad-jlcpcb-tools-main/mainwindow.py", line 18, in <module>
    from .fabrication import Fabrication
  File "/Users/akelly/Documents/kicad/6.0/scripting/plugins/kicad-jlcpcb-tools-main/fabrication.py", line 8, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
```

I ran `/opt/local/Library/Frameworks/Python.framework/Versions/3.10/bin/python3 -m pip install requests` and that fixed it

Bouni, 2022-10-24:
Glad you firgured it out 🥳

closed 2022-10-24 by Bouni