Bouni/kicad-jlcpcb-tools #234: Project status info

Bouni, 2022-10-31:
Hi everyone,

I guess many of you realized that this projects pace slowed down dramatically over the last two months or so.
Thats du to some events happening in my family (nothing bad, don't worry 😉 ) which needs much of my spare time.

For me its always family first, so the project has to be lower in priorty and you guys hopefully understand that.

I'll try to merge PRs as soon as possible and answer and probably solve issues quickly but I cannot guarantee that 🤷🏽 

But I'm sure that over time I'll get back to work on improvements and new features!

hedon77, 2022-11-05:
Congratulations.. Best Regards

chmorgan, 2025-03-29:
Closing due to inactivity.