Bouni/kicad-jlcpcb-tools #238: Tool not working in KiCAD 6.0.9 on MacOS

BenViete, 2022-11-08:
**Describe the bug**
After installing the plugin (plugin manager / GitHub clone) I try to start the plugin. If there is no jlcpcb folder in my project folder it starts, but has no parts listet on the popup window. If I close the window and try to start the plugin again I get the following error message:

```
Traceback (most recent call last):

File "/Users/vieteben/Documents/KiCad/6.0/scripting/plugins/kicad-jlcpcb-tools/plugin.py", line 21, in Run
dialog = JLCPCBTools(None)

File "/Users/vieteben/Documents/KiCad/6.0/scripting/plugins/kicad-jlcpcb-tools/mainwindow.py", line 571, in __init__
self.init_store()

File "/Users/vieteben/Documents/KiCad/6.0/scripting/plugins/kicad-jlcpcb-tools/mainwindow.py", line 587, in init_store
self.populate_footprint_list()

File "/Users/vieteben/Documents/KiCad/6.0/scripting/plugins/kicad-jlcpcb-tools/mainwindow.py", line 660, in populate_footprint_list
details = self.library.get_part_details(numbers)

File "/Users/vieteben/Documents/KiCad/6.0/scripting/plugins/kicad-jlcpcb-tools/library.py", line 318, in get_part_details
return cur.execute(

sqlite3.OperationalError: no such table: parts
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install plugin in KiCAD 6.0.9 on MacOS
2. Click on JLCPCBTools button
4. See error message (exception)

**Expected behavior**
Tool should run as expected.

# KiCad Version
```
Application: KiCad Leiterplatteneditor
Version: (6.0.9-0), release build
Libraries:
    wxWidgets 3.1.5
    libcurl/7.79.1 SecureTransport (LibreSSL/3.3.6) zlib/1.2.11 nghttp2/1.45.1
Platform: macOS Version 12.6.1 (Build 21G217), 64 bit, Little endian, wxMac
Build Info:
    Date: Oct 29 2022 19:31:49
    wxWidgets: 3.1.5 (wchar_t,wx containers)
    Boost: 1.80.0
    OCC: 7.6.3
    Curl: 7.64.1
    ngspice: 37
    Compiler: Clang 12.0.0 with C++ ABI 1002
Build settings:
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON
```

Bouni, 2022-11-25:
Can you please check if there's a `parts.db` file in `/Users/vieteben/Documents/KiCad/6.0/scripting/plugins/kicad-jlcpcb-tools/jlcpcb/`?
If so, delete it it and see if the plugin starts.

binary1230, 2022-11-26:
I am hitting a similar (same?) issue after upgrading my plugin from the august version to today's version (nov 11th or something like that).

[image]

seeing this go by in the error console too.
```--- Logging error ---
Traceback (most recent call last):
  File "C:\Program Files\KiCad\6.0\bin\Lib\logging\__init__.py", line 1086, in emit
    stream.write(msg + self.terminator)
AttributeError: 'NoneType' object has no attribute 'write'
Call stack:
  File "C:\Users\Dominic Cerquetti\Documents\KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\plugin.py", line 21, in Run
    dialog = JLCPCBTools(None)
  File "C:\Users\Dominic Cerquetti\Documents\KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\mainwindow.py", line 571, in __init__
    self.init_store()
  File "C:\Users\Dominic Cerquetti\Documents\KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\mainwindow.py", line 585, in init_store
    self.store = Store(self.project_path)
  File "C:\Users\Dominic Cerquetti\Documents\KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\store.py", line 30, in __init__
    self.update_from_board()
  File "C:\Users\Dominic Cerquetti\Documents\KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\store.py", line 224, in update_from_board
    self.logger.debug(
Message: 'Part D1 is already in the database and has a lcsc value, the value supplied from the board will be ignored.'
Arguments: ()
2022.11.25 22:30:23 - DEBUG - update_from_board -  Part D1 is already in the database and has a lcsc value, the value supplied from the board will be ignored.```

I'm on Kicad 6.0.6 though on Win64.

```Application: KiCad PCB Editor (64-bit)
Version: (6.0.6), release build
Libraries:
    wxWidgets 3.1.6
    libcurl/7.82.0-DEV Schannel zlib/1.2.12
Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Date: Jun 26 2022 02:39:06
    wxWidgets: 3.1.6 (wchar_t,wx containers)
    Boost: 1.79.0
    OCC: 7.6.0
    Curl: 7.82.0-DEV
    ngspice: 37
    Compiler: Visual C++ 1929 without C++ ABI
Build settings:
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON```

I was about to open this up on my PC and dig in a little.  

One possible hint, but it might be unrelated: I saw an error message in the previous version of the plugin when trying to hit the "Update Database" button that the JLCPCB URL was returning a 404.  wonder if that has something to do with this.

I'll try remove that parts.db file if it exists as well.

binary1230, 2022-11-26:
I'm on plugin verison 2022.11.01 with this

binary1230, 2022-11-26:
ok, so I did have a parts.db file at 
C:\Users\Dominic Cerquetti\Documents\KiCad\6.0\3rdparty\plugins\com_github_bouni_kicad-jlcpcb-tools\jlcpcb\parts.db

when I opened it in DB browser, I see there is only an table in it named 'mappings' (and no table named 'parts')

when I delete the file, I am able to get the plugin to open, but, now i get the 404 error.  I'll try chasing that down next.

[image]

binary1230, 2022-11-26:
If I change the line in library.py to print the URL, I get this:
[image]

https://jlcpcb.com/componentSearch/uploadComponentInfo

binary1230, 2022-11-26:
yep,
[image]

and then if I close the plugin I get the parts.db thing happening again and have to delete it.  I think there's 2 things going on here then:
1. The plugin doesn't handle itself well when it wasn't able to previously reach JLCPCB and (presumably populate the db).
2. (root cause) the JLCPCB url seems to have legit changed.

I'll see if I can track down the new URL, hopefully they didn't block automated API requests like this plugin or something.

binary1230, 2022-11-26:
wonder if the base URL changed. this now works:
https://jlcpcb.com/parts/componentSearch/

sadly https://jlcpcb.com/parts/componentSearch/uploadComponentInfo does not.

searching....

binary1230, 2022-11-26:
oh uh, just noticed https://twitter.com/bouni2016/status/1453258658545115141 , probably similar thing going on

Bouni, 2022-11-26:
Seems JLC removed the download link from their website. Hopefully they just moved it elsewhere, otherwise a good portion of this plugin is useless 😥

Bouni, 2022-11-26:
Maybe related to #241

andylinpersonal, 2022-11-28:
[image]
They cannot provide the link to the database now 👎🏻

binary1230, 2022-11-28:
Translation from Google translate
```
At present, we do not provide this file for downloading, it is recommended that you search directly in the component library```

BenViete, 2022-11-28:
I think I don't need to do anything, because the problem is described above.

The parts.db file is available in my folder. If I delete it, then the plugin starts. But there is no part available.

binary1230, 2022-11-28:
Hey all, I suggest we take any future discussion into #241 , which is the root cause of this issue. (Just to keep it all in one place)

Bouni, 2022-12-06:
I'll close this. See #241 for details.

closed 2022-12-06 by Bouni