Bouni/kicad-jlcpcb-tools #239: Export GUI Stock Levels to CSV

EShirkIoT, 2022-11-24:
Fairly straight forward, it would be nice to be able to easily work with the stock level values in a spreadsheet to accommodate for alternatives if needed.

Bouni, 2022-11-25:
Maybe a general export for the table shown in the GUI would be nice. I just want to avoid cluttering the GUI with dozens of buttons 😅 Maybe get this function from the right click menu!?

What do you think?

EShirkIoT, 2022-11-25:
Sounds good, just would be really handy when trying to compare stock levels and alternatives. 
When you say right click menu, do you mean like when you right click an item in the table?
Honestly as long as its possible I'm happy :)

z2amiller, 2026-09-20:
BTW - all of this info is available in the sqlite database that ships -- so you can do complex SQL queries to pull whatever info you'd like from the parts database.  Really, using the parts search pane is probably the right place to investigate alternate components with better stock levels though.

closed 2026-09-20 by z2amiller