Bouni/kicad-jlcpcb-tools #241: Not Able to Download JLC Library Database (404)

abish7643, 2022-11-26:
**Describe the bug**
I tried to update to the latest version of kicad-jlcpcb-tools, upon using it, there's an error as shown in the image.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to PCB Editor
2. Click on the Plugin button
3. Click on Update Library

**Expected behavior**
Automatically downloading LCSC Part Library.

# KiCad Version

```
Application: KiCad

Version: 6.0.9-8da3e8f707~117~ubuntu22.10.1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.85.0 OpenSSL/3.0.5 zlib/1.2.11 brotli/1.0.9 zstd/1.5.2 libidn2/2.3.3 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.49.0 librtmp/2.3

Platform: Linux 5.19.0-23-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Nov 24 2022 00:22:25
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.6.3
	Curl: 7.85.0
	ngspice: 37
	Compiler: GCC 12.2.0 with C++ ABI 1017

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

abish7643, 2022-11-26:
[image]

Bouni, 2022-11-26:
404 means that JLC might changed the download link to their database.
I'll take a look on Monday

abish7643, 2022-11-26:
Makes sense. Thanks for the quick response.

binary1230, 2022-11-26:
yea, def the root cause of #238 , I looked around and couldn't find an equivalent CSV download, though, I don't really know where to look.

in a pinch, I would love to use the rest of the plugin even without the stock levels or parts data, just because the POS/BOM export/etc is still really useful.  

I'm having a look to see if we can catch some of the sql exceptions and continue on even if the URL 404's out (unless you think that's just a silly thing to attempt)

binary1230, 2022-11-26:
yea ok I think i have some changes that help restore at least the ability to export and use the plugin, will prep a PR

elementzonline, 2022-11-26:
Same issue here. 
Wondering if I can get an old CSV and use that for the time being.

binary1230, 2022-11-26:
OH right. you know what..... maybe by a stroke of luck i might have a copy of the parts.db (built from the CSV) from the last download I did a while ago in my nightly backups. lemme check....

if anyone has more recent CSV dumps though (somehow), definitely post em

there's also a few other kicad+JLC related projects that do similar things, wonder if anyone has some kind of new info about what the best way to do this is.  I notice https://github.com/yaqwsx/jlcparts/issues/78  was just reported (looks like this issue started happening 5 days ago then)

CC'ing @yaqwsx just in case there's anything we can do to help with reaching out to JLCPCB/LCSC

last build up there was: https://github.com/yaqwsx/jlcparts/actions/runs/3520139755 for reference for the cached states (if using it is OK and/or useful)

Bouni, 2022-11-26:
On Monday I'll try to modify the plugin so that it at least works without the CSV/parts.db.
Maybe I can find a nice way to reverse engineer the API used by JLCs /parts webpage and use that ...

binary1230, 2022-11-26:
@Bouni I threw in a couple PRs, the main one that matters is https://github.com/Bouni/kicad-jlcpcb-tools/pull/242/ I threw in just enough exception handling (and some hackery) to let the plugin start so you can still do exports.  beyond that, it's a fix that could do with some more help.

binary1230, 2022-11-26:
also, here's a copy of a restored parts.db (~500mb) from Nov 17th from a backup from my laptop, if it's useful.

Perhaps we can code up some kind of option to use this file as a temporary "offline cache" that the plugin can use to at least startup a bit more normally.

Go easy on my bandwidth :)   requires git LFS to download (or might work OK from the github webui)
https://github.com/binary1230/offline-jlcpcb-kicad-tools-data/blob/main/backups/offline-cached-partsdb-kicad-jlcpcb-tools--11-17-2022.parts.db

yaqwsx, 2022-11-26:
@Bouni LCSC (and JLC PCB, I guess) plan a huge internal rework of their systems on Dec 3rd. This might be related. I emailed them a few days ago but without any response. Unfortunately, I don't keep a copy of the CSV table on my service.

andylinpersonal, 2022-11-28:
[image]

Currently, they don't provide the link to the database :/

andylinpersonal, 2022-11-28:
[image]
They don't provide the link to the database now 👎🏻

binary1230, 2022-11-28:
See in #238 , like like they're saying the csv won't be available and they recommend "searching the website" instead. Boo.

elementzonline, 2022-11-28:
It's a shame that cannot provide a catalog file. 
I came a across the following link for Api. Not sure if it still works 

https://github.com/yaqwsx/jlcparts/blob/master/LCSC-API.md

As per the Readme file it says we will be getting Json response for the product part number.

Bouni, 2022-11-28:
I just released a emergency fix (hack) that let you genereate producton files but the search for parts no longer work ...

hihey1, 2022-11-28:
> 

But when I click update library, I still get a 404 error and the search function is disabled.

Bouni, 2022-11-28:
@hihey1 That's because JLCPCB removed the download link for the database. This is an emergency hack so that you can at least use the plugin to generate production files!

Bouni, 2022-11-28:
A `parts.db` from 2022-11-23 can be found here: https://jlc.bouni.de/parts.zip.
If you extact tha into 
 - Windows: `C:\Users\<username>\Documents\KiCad\<version>\scripting\plugins\kicad-jlcpcb-tools\jlcpcb`
 - MacOs: `/Users/<username>/Documents/KiCad/<version>/scripting/plugins/kicad-jlcpcb-tools/jlcpcb`
 - Linux: `/home/<username>/.local/share/kicad/<version>/scripting/plugins/kicad-jlcpcb-tools/jlcpcb`

You should be able to use all plugin functions with that expet for download a newer version obviously.

hihey1, 2022-11-29:
> 
After I extract the parts.db, all functions can run normally, but after I run it once again, there is a 404 error still.

leecyrille, 2022-11-29:
[image]
Even with the parts database manually added I can't seem to search anymore :(

Really hope JLC doesn't break this tool! (unless they break it in the process of creating a functional search of their own?)

Bouni, 2022-11-29:
> but after I run it once again, there is a 404 error still.

Did you install the latest version of the plugin? [2022.11.02](https://github.com/Bouni/kicad-jlcpcb-tools/releases/tag/2022.11.02)

hihey1, 2022-11-29:
> 

Yes, I have installed the latest version of the plugin, but I can only seem to run the plugin once at a time, and the next time I want to search again, I have to uninstall and download the plugin again before it works.

pico01, 2022-11-29:
at my Win11 PC the Path for the parts.db is
C:\Users\[use]\Documents\KiCad\6.0\scripting\plugins\kicad-jlcpcb-tools-main\jlcpcb
mabe this helps

Bouni, 2022-12-02:
The fallback is mergend and released with 2022.12.01

Bouni, 2022-12-06:
🥳 Partial success!

JLC added the CSV downlaod link to my account (I see it where it was when logged in).
That means I can manually download the file and generate the parts.db from it.

I think its not possible to automate this at this point because of recaptcha etc. but I sent a request to them and asked for a way to allow me to download it automated once per day. 

Maybe they agree and give me some kind of token based access to the resource. If thats the case I'l build a small web service that fetchs the CSV once per day and builds the parts.sb from it. That way we can leave the plugin in the current state and download the zipped parts.db instead of the CSV.

Bouni, 2022-12-06:
:unamused: As expected they don't make the file available via token or so.

No is the big question, how can we maintain a up to date parts.db?

I'm sure I cannot download the file manually every day so I maybe contiue to work on a crawler that gets all the parts from jlcs website / public API.

Any thoughts?

Bouni, 2022-12-06:
@yaqwsx 😄 I was about to write you an email and discuss how we can get this working again 

Their answer was clearly though:

We are sincerely sorry we really want to help, but we do cannot help with this.

> Bouni: I cannot login manually every day to download the file, so I'll build a solution that gets the file for me every day, but therfore I need a way with a token for example that allows me to download the file.

> JLC: In fact it is already a privilege and currently only your account can download it. Since we would have stopped offering the catalog and therefore can no longer do the update. All we can do is keep the function for you. 

Any suggestions how to apply a little more preassure so that they might change their minds?

binary1230, 2022-12-06:
Cc @cnlohr : just FYI

binary1230, 2022-12-06:
I posted a link to this in a slack channel I'm in that might be able to get some attention going, if you guys feel that's useful and would know how/ where to direct it.

I suggest if you're going to go for something like that, request everyone to coordinate whatever public thing (a tweet or whatever) at a specific date and time to maximize the impact.

DomoticX, 2022-12-06:
> 😒 As expected they don't make the file available via token or so.
>
> No is the big question, how can we maintain a up to date parts.db?
>
> I'm sure I cannot download the file manually every day so I [...]

Is this feature only created for your account?, or can everybody that has a JLCPCB account download this file? (the people using this plugin obvisously have a JLCPCB account!), so you could put in some input text fields to enter user/pass in the tool, and use the API (HTTP GET/POST) to get the token based on useraccount?.. and download the file? just an idea...

Bouni, 2022-12-07:
@DomoticX Yes, as far as I can tell they only added the download link back to my account (I see it in the top right corner of the parts table where it used to be).
I already tinkered with automated downloading but the login is secured by recaptcha which makes it (at least for me) impossible to automate the login required for the download.

Bouni, 2022-12-07:
**Update:**

After my [tweet](https://twitter.com/bouni2016/status/1600078603462508545) JLCPCB reached out to me once agian and asked me to contact them via email and describe to them what I mean by "token based access".

So maybe they will eventually give me a way of automatically download the CSV, lets hop for the best.

**Plan B:**

I once more tinkered with their public API (the one they use in https://jlcpcb.com/parts/).
It seems that I could use that directly in the plugin, so we no longer need to have the CSV.
There are still a few more test to be done but it looks not that hard to use.
After I'm done with my test, I'll publish my findings.

elementzonline, 2022-12-11:
Plan B seems to be the way to go considering that they are not providing the CSV anymore. Good luck with that. I wish I could help with some part of the development as well. Really loving this plugin. Thanks for making it available for makers like us.

MichaelRColton, 2022-12-15:
Yikes! Is there a way we can reach out to them as a group to get them to reconsider? I literally switched to using JLCPCB's assembly service _**because of this plugin**_ and it just won't be worth it to me to make as many boards if I can't automate this part of the process.

MichaelRColton, 2022-12-15:
I read too fast, I missed the potential Plan B. That does sound better than having a batch that you have to download and process and provide to us each day (if I'm understanding that process correctly) I'll be excited to hear how that goes.

Bouni, 2022-12-15:
**Update:** I was in touch with JLC again and the promised to come up with a solution (public API as far as I can tell) that will allow tools like this to use their parts library directly 🥳 

Bt they couldn't give me a deadline when they finish this. Anyway, nice that the move in this direction 😄

MichaelRColton, 2022-12-15:
That's great news. Fingers crossed that it happens soon. I also understand that you are busy. We appreciate the great work you are doing!

binary1230, 2022-12-15:
https://media.tenor.com/8d419rqb5qsaaaac/airplane-goodluck.gif

binary1230, 2022-12-15:
[image]

zbyhuu, 2022-12-19:
@Bouni
How to generate db file from locally downloaded csv file? 
Did you considered to add a new button with "upload  csv file manually" function instead of autodownload from JLC?

Bouni, 2022-12-19:
@zbyhuu Here is where the magic happend: https://github.com/Bouni/kicad-jlcpcb-tools/blob/2022.09.01/library.py#L324

From where should that CSV come from? Currently my account is the only one that can still download the CSV so the button is pointless.

zbyhuu, 2022-12-19:
@Bouni 
CSV could come from your local disk.
JLC enabled "download table" button for me also, so for WA this issue now, creating the *.db file from local *.csv file could help... at least to JLC customers who request such access.

Bouni, 2022-12-29:
**Update:** 🥳 

JLC send me their first API draft. It's not very useful at the moment as it only allows me to fetch teh first 1000 parts from their database but I already replied to them with suggestions for improvements.
It seems that we're on the right track towards a real API integration into this project!

My plan is to implement the API into a seperate python module so that it can be used by other projects as well.

Give me and JLC some time to fully work this out :blush:

binary1230, 2022-12-29:
Awesome! That's great to hear! Always cool when the actual company works with the maintainers.

MichaelRColton, 2023-03-02:
I hate to bother you, but any update on this? Thanks for all you are doing!

Bouni, 2023-03-02:
Unfortunately no. JLC contacted me in late January and said they will get in touch with me regarding their new API but never did that.

Honestly I also totally forgot about that. I'll write them an email and see what they have for us

DanielO, 2023-03-17:
I had a chat with @yaqwsx and they said we could piggy back off their API key if we supply a GH CI job.

I have written a script which extracts a suitable CSV file from the jlcparts cache.db which compresses down to about 27MB which can be hosted on GitHub.

Bouni, 2023-03-17:
@DanielO Nice! Send a PR for GH Action that runs ever night or so and I'll merge it!

DanielO, 2023-03-19:
It needs an API key I don't have but @yaqwsx seemed receptive to piggy backing off the jlcparts project to generate the CSV we need, I am chatting via email about it.

yaqwsx, 2023-03-20:
What I suggest to @DanielO was to spin up a custom GH actions job that pulls the DB from JlcParts, builds the original CSV table and hosts it via GH pages.

MichaelRColton, 2023-03-21:
Glad to see potential progress on this!

DanielO, 2023-03-21:
Oh I see what you mean now, that does make more sense 😂

Bouni, 2023-03-24:
@yaqwsx Am I reading [your code ](https://github.com/yaqwsx/jlcparts/blob/master/jlcparts/jlcpcb.py) right that you iterate over all pages of the parts API in order to get the data?

I last heard from JLC on the 29th of december when they promised me a swagger like documentation and additional features, but since then there was silence (except one DM on twitter with the promise of more info soon ..)

I wrote them an email just minutes ago as a reminder, lets see what comes from that ...

DanielO, 2023-03-25:
OK I finally got around to setting up a job, see https://github.com/DanielO/kicad-jlcpcb-tools/blob/csv-creation/.github/workflows/update_csv.yaml

Note that my repo also has fixes for KiCAD 7 but I haven't tried working on backwards compatibility yet..

DanielO, 2023-03-27:
I updated it to have KiCAD6 compatibility, I think those changes should probably be cherry picked out separately but having them all in one place seems easier for testing.
One problem is that KiCAD 7.0.0 seemed very crash happy while downloading, I reduced the frequency of logging/widget updates and it seems better, there is also version 7.0.1 which may have helped.

markusdd, 2023-05-08:
mmh @Bouni I would think that I actually solved this with #307 and #310

Bouni, 2023-05-09:
This is fixed by the awesome work of @markusdd and the help of @yaqwsx who provides the data source 🥳

closed 2023-05-09 by Bouni

zbyhuu, 2023-05-10:
Much appreciated!