Bouni/kicad-jlcpcb-tools #248: It's not possible to sort the table on the "JLCPCB Tools" main window by clicking on the column titles

moskothebossko, 2022-12-08:
It's not possible to sort the table on the "JLCPCB Tools" main window by clicking on the column titles.   

Setup:

Application: KiCad PCB Editor

Version: 6.0.9+dfsg-1~bpo11+1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.74.0 OpenSSL/1.1.1n zlib/1.2.11 brotli/1.0.9 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.43.0 librtmp/2.3

Platform: Linux 5.10.0-19-amd64 x86_64, 64 bit, Little endian, wxGTK, gnome, wayland

Build Info:
	Date: Nov  3 2022 21:14:23
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.1
	Curl: 7.74.0
	ngspice: 34
	Compiler: GCC 10.2.1 with C++ ABI 1014

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

Cheers 
Daniel

Bouni, 2022-12-12:
I'v just tested with 6.0.9 on Windows an here it wors:

```
Application: KiCad Leiterplatteneditor (64-bit)

Version: (6.0.9), release build

Libraries:
	wxWidgets 3.2.1
	libcurl/7.83.1-DEV Schannel zlib/1.2.12

Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Oct 29 2022 23:44:28
	wxWidgets: 3.2.1 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.0
	Curl: 7.83.1-DEV
	ngspice: 37
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

I fear that this is again a issue with the wxwidgets version packaged by KiCAD. That differs from platform to platform which really sucks and case on headache after the other for me 😞

z2amiller, 2026-04-10:
I think this is obsolete - it's worked on every OS I've tried (only windows and macos, sorry).
I'm going to close this but please re-open if this is still an issue on your OS.

closed 2026-04-10 by z2amiller