Bouni/kicad-jlcpcb-tools #249: update polygon pour even if cancel the repour operate

sonnyyin, 2022-12-09:
**Describe the bug**
Hi, I use kikit plugin for panelize a pcb board, then export gerber use JLCPCB-tool, when I click [Generate fabrication files] button, a tipbox show and ask for repour polygn, even if I click the [cancle] button , the polygon still update. This result copper conncet together.

**To Reproduce**
Steps to reproduce the behavior:
[image]

the result image:
[image]

**Expected behavior**
Dont't update the polygon area.

# KiCad Version

```
Application: KiCad PCB 编辑器 (64-bit)

Version: (6.0.9), release build

Libraries:
	wxWidgets 3.2.1
	libcurl/7.83.1-DEV Schannel zlib/1.2.12

Platform: Windows 10 (build 19042), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Oct 29 2022 23:44:28
	wxWidgets: 3.2.1 (wchar_t,wx containers)
	Boost: 1.79.0
	OCC: 7.6.0
	Curl: 7.83.1-DEV
	ngspice: 37
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

Bouni, 2022-12-12:
This seems like a KiCAD bug. The modal is generated by KiCAD and should handle the yes / no user choice. I'll file a bugin KiCAD's Gitlab and see if I'm right.

Bouni, 2022-12-12:
KiCAD issue for reference: https://gitlab.com/kicad/code/kicad/-/issues/13135

sonnyyin, 2022-12-12:
Thanks for reply，I will focus on it.

Bouni, 2023-05-26:
Fixed in #314

closed 2023-05-26 by Bouni