Bouni/kicad-jlcpcb-tools #250: Rotations need overhaul

Bouni, 2022-12-12:
If there are two rotations defined in the Rotation manager that both match, for example `^SOT-363` and `^SOT\-363_SC\-70\-6` then both are applied and just the last will have an effect. This might lead to an unwanted result.

Furtermore not all packages need the same rotation correction. So it might be smarter to have two options, per package correction and per part name correction where the later has priority.

dzid26, 2022-12-17:
Could rotations also be exported to the footprint field?  Preferably, using one of the field names from another [plugin ](https://github.com/bennymeg/JLC-Plugin-for-KiCad/pull/21/files#diff-fbc945fd5da91f03cd17c251f9255b7c6f06ef3030148ba3e3e244f248e38069R195-R196) for compatibility.

Bouni, 2022-12-19:
Defenitly something to consider. The only problem at the moment is that footprint values get erased whenever you sync with your schematic. There are plan from the KiCAD folks to add an API that allows back writing into the schmeatic but thats planned for V7 ....
Also we have [kind of a write back](https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/schematicexport.py) but its a bit of a hacky and risky procedure in my eyes

Bouni, 2022-12-30:
I added a better rotation correction method in #254 

@dzid26 Your request wil be deferred until KiCAD has a real API to write into the schematic.

closed 2022-12-30 by Bouni