Bouni/kicad-jlcpcb-tools #251: Problem repository link

okhanvatansever, 2022-12-29:
**Describe the bug**
Problem installing Kicad PCM. Error with repository link: "Failure when recieving data from the peer" "could not read repository url".

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Plugin and content manager and add new repository link.
2. See error

**Expected behavior**
Normally I would have added the repository so I could install the addon,

# KiCad Version
6.0.10

Bouni, 2022-12-29:
@okhanvatansever I've just upgraded from 6.0.9 to 6.0.10 and suddenly I have the same issue:

[image]

I guess this is some KiCAD issue. Let me check if there are open issues on KiCad's Gitlab.

Bouni, 2022-12-29:
I filed a new issue on KiCADs GitLab: https://gitlab.com/kicad/code/kicad/-/issues/13307

Bouni, 2023-01-30:
Fixed in Kicad `6.0.11` and `7.0.0-rc2-62-g971abcfc1a`

closed 2023-01-30 by Bouni