Bouni/kicad-jlcpcb-tools #256: Release 2022.12.04 does not show up in PCB editor, but release 2022.12.02 does work

Athanaze, 2022-12-31:
I tried the release 2022.12.04 on two Linux machine (Debian KDE and manjaro i3), and on windows 11.
In all platform, after installing (I tried via the gui and with putting the files manually in the plugin folder), after closing and restarting kicad, the jlcpcb button does not appear in the tool bar in the pcb editor.

It does not appear in Tools > external plugins either.

I installed the previous release, 2022.12.02, and it works fine.

agalliazzo, 2022-12-31:
Same for version 2022.12.03 and cloning directlry from git, windows 11 and KiCad 6.0.9... I can confirm latest working version is 2022.12.02

Bouni, 2022-12-31:
Can you check settings -> action plugins -> ⚠️ button. It might show what's the cause

florian0, 2023-01-01:
I tested 2022.12.04 (via Plugin Repo) on KiCad 6.0.9 on Linux.
```
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/florian0/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/__init__.py", line 4, in <module>
    from .plugin import JLCPCBPlugin
  File "/home/florian0/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/plugin.py", line 5, in <module>
    from .mainwindow import JLCPCBTools
  File "/home/florian0/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 18, in <module>
    from .fabrication import Fabrication
  File "/home/florian0/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/fabrication.py", line 9, in <module>
    from pcbnew import (
ImportError: cannot import name 'DRILL_MARKS_NO_DRILL_SHAPE' from 'pcbnew' (/usr/lib/python3.10/site-packages/pcbnew.py)
```
Looks like DRILL_MARKS_NO_DRILL_SHAPE is only available in Kicad nightly right now. That's why its not working.

Bouni, 2023-01-01:
Ok I know what causes the problem but I can't fix it before Tuesday 🫤

AdamSBlack, 2023-01-01:
> Ok I know what causes the problem but I can't fix it before Tuesday 🫤 As a quick fix for 6.0 users you can comment this line
> 
> https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/fabrication.py#L10

This worked for me, cheers.

Bouni, 2023-01-02:
I hope that #257 will fix this, had no chance to test it on a 6.0 installation 😅

agalliazzo, 2023-01-02:
Yep, it worked for me! Thanks for your efforts! :)

closed 2023-01-02 by Bouni