Bouni/kicad-jlcpcb-tools #260: Seperate rotations from parts datasbase

Bouni, 2023-01-04:
**Is your feature request related to a problem? Please describe.**

rotations are currently saved in the same database as the parts. So they get deleted when a new db version is downloaded.

**Describe the solution you'd like**

We should put them into a seperate database.

**Additional context**

Also a migration for existing rotatiosn should kick in when this feature is added.

skuep, 2023-01-11:
Was just hit by this. Is it possible to bake the rotation into the project database mappings? It kind of seems really bad to me, that generation of production files is dependent on what computer I run it on (and the state of its local database). I would want to configure this once for a project, and it should stay like this (for this project).

skuep, 2023-01-14:
I think that would be the optimum solution. I have not yet checked deeper into your plugin, but I noticed that right now, it already does things like for example this:
``2023.01.14 09:45:30 - DEBUG - update_from_board -  Part R14 is already in the database and has a lcsc value, the value supplied from the board will be ignored.`` when opening the LCSC plugin window.

Would it be possible, lets say to add to the project.db a column "rotation offset" for each part on the board, which is NULL by default. Now, when opening the LCSC plugin window, it (not only does the Part update as shown above but also) checks if there are rotation offsets, that are NULL for specific parts (or footprints). Then it will perform a database lookup and set a value, if it finds the footprint. If no database entry is found, the value is set to zero (if you would leave it at NULL, the next guy with a different database might add potentially bogus rotation offsets).

This way, you can use the database for unitialized rotation offsets once, but also overwrite and set a stable project-constant value that does not change without expecting it.

Bouni, 2023-01-14:
I'll think about over the weekend and let you know on Monday what I came up with.
Thank you for describing your thoughts so detailed, appreciate it 👍

computergeek1507, 2023-01-14:
If you make a rotation database can you have it include X/Y position offsets? Maybe call it the "adjustment" database or similar. I want to add X/Y offsets property for through holes that don't line up correctly.

Bouni, 2023-01-14:
@computergeek1507 good point, definitely going to include that 👍

DomoticX, 2023-01-28:
> If you make a rotation database can you have it include X/Y position offsets? Maybe call it the "adjustment" database or similar. I want to add X/Y offsets property for through holes that don't line up correctly.

I have made this request over here:
[https://github.com/Bouni/kicad-jlcpcb-tools/issues/201](https://github.com/Bouni/kicad-jlcpcb-tools/issues/201)

closed 2023-02-07 by Bouni

skuep, 2023-02-07:
Very cool, thanks a lot!

skuep, 2023-06-07:
Playing around with your Plugin again and I was under the impression that with your changes, I could store the part rotations into the project.db file. Looks like that feature never made it into the Release? :-D

The basic problem I have is, that if, let's say, I get a new computer and checkout my KiCad project, the part rotations are all wrong, because I do not have the rotation override mappings installed onto my computer. Thus, the idea was to hard-bake the part rotations into the project.db file, so that whoever opens my project, they have all the rotation offsets already there (and possibly overriding their local rotation offsets they might have installed in the system). I.e. if the project.db says "NULL" as part rotation, use the system file. Otherwise use the value in project.db

Is something like that possible? Best Regards, I really love your plugin! :-)