Bouni/kicad-jlcpcb-tools #261: Show LCSC details in parts table

Bouni, 2023-01-04:
**Is your feature request related to a problem? Please describe.**

When changing a component value, a 1k resistor for a 10k for example it could easily happen that one does not update the LCSC  number.
On the Plugin it then looks like the right LCSC number is assigned which results in JLC populating the wrong part.
That happened to me twice 😓 

**Describe the solution you'd like**

Ideally we show the LCSC descrption in the parts tabel along with the component value so we could visually see if the selected part is correct.

Bouni, 2023-01-04:
We read the LCSC value from the schematic which is where you ideally want to select a part. Unfortunately the API does not yet allow for interaction with the schematic but that is scheduled for v7.

I'm not sure if it's worth the effort to check a change in value and LCSC 🤔

I also would like to have a check if I not accidentally put the wrong LCSC number in the schematic (100k resistor instead of a 10k or so)

z2amiller, 2026-09-26:
This should be closed now - there is highlighting in the mainwindow that shows matches for value/footprint, and when a value/footprint changes on the board we clear the LCSC.

closed 2026-09-26 by z2amiller