Bouni/kicad-jlcpcb-tools #262: "Toggle BOM" / "Toggle POS" / "Toggle BOM/POS" makes the table jump to the top

deragent, 2023-01-05:
**Describe the bug**
In the main KiCAD-JLCPCB Window, one can toggle the inclusion of parts in the BOM / POS via 3 buttons (mentioned in the title).

Pressing any of these buttons correctly toggles the BOM resp. POS flags, but also makes the table scroll back to the top.
This also happens when selecting an LCSC part (via the "Select part" button).

This is of course not a game-changer, but rather annoying when working with boards with a large amount of parts.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the "JLCPCB Tools" in pcbnew
2. Scroll all the way down to the bottom of the parts table.
3. Select the any line at the bottom.
4. Press "Toogle POS"
5. Correct: The POS flag is toggled.
6. Bug: The table jumps back to the top.

**Expected behavior**
The table should not scroll back to the top, but stay at the current location.

# KiCad Version

```
Application: KiCad PCB Editor

Version: 6.0.10+dfsg-1~bpo11+1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.74.0 OpenSSL/1.1.1n zlib/1.2.11 brotli/1.0.9 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.43.0 librtmp/2.3

Platform: Linux 4.19.0-16-amd64 x86_64, 64 bit, Little endian, wxGTK, , tty

Build Info:
	Date: Dec 31 2022 08:48:09
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.1
	Curl: 7.74.0
	ngspice: 34
	Compiler: GCC 10.2.1 with C++ ABI 1014

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

deragent, 2023-01-05:
The source of this bug is, that the parts list / table is recreated after any action.

https://github.com/Bouni/kicad-jlcpcb-tools/blob/ab6b2151e5b62b5ed9c8a9d197f77b19a9bb46e5/mainwindow.py#L618

As part of this, the table is first cleared and the filled again with the entries.

Solving this properly would probably involve a larger rewrite, as the it would mean, that the table is not cleared on any action, but that the result of any action is selectively applied to the concerned rows, without clearing them.

I am wondering, if (as a more simpler fix) it is possible to extract the current scroll position before clearing the table, before

https://github.com/Bouni/kicad-jlcpcb-tools/blob/ab6b2151e5b62b5ed9c8a9d197f77b19a9bb46e5/mainwindow.py#L622

and then to manually reset the same scroll position once the table has been refilled, after

https://github.com/Bouni/kicad-jlcpcb-tools/blob/ab6b2151e5b62b5ed9c8a9d197f77b19a9bb46e5/mainwindow.py#L681

Not sure if this is possible with wx?

deragent, 2023-01-05:
After a quick search, at least in 2014 it was not possible to get the exact scroll position of a `wxDataViewCtrl` (see: https://discuss.wxpython.org/t/dataviewctrl-how-to-scroll-programmatically-wxgtk/28935/8). I was not able to find if this has changed in the meantime.

But there seems to be a workaround, by using `wxDataViewCtrl::EnsureVisible` as mentioned in the linked post above.

This could be used, to at least make sure, that the currently selected item is still visible (in the visible scroll range) after the table recreation. This would probably still lead to some jumps, but to a much lesser degree.

Bouni, 2023-01-05:
Seems to be a linux only issue, on windows no jump at all ...

[image]

But thats just a guess at the moment, I'll verify this hopefully tomorrow ....

deragent, 2023-01-05:
I wonder if it might also be a wx-Version issue? Bellow a screencast of the behavior in my case: 

[image]

For completeness: I am using JLCPCB-Tools v. 2023.01.01.

wrs, 2023-06-20:
FYI the scroll-to-top also happens on MacOS (tools 2023.05.03).

Bouni, 2023-06-20:
@wrs @deragent Would you please test my PR #322 and check if that fixes the jump issue?

closed 2023-06-29 by Bouni