Bouni/kicad-jlcpcb-tools #267: Scripting/Running Jlcpcb-tools from command line

hattesen, 2023-01-16:
# running JLCPCB-tools from command-line

This feature may already be present in the current (Python) implementation of the plug-in, but I have not been able to find any information about it. 

I would like to be able to control the functions of this KiCAD plugin from the command line (e.g. by calling the script directly from the command line with command line arguments) to be able to automate the generation of documentation and production files for a project. This would not only allow consistent generation of files every time a new version of the schematic/PCB is made, but these files could be automatically generated by a CI server (e.g. GitHub workflow/actions, Travis, CircleCI, Jenkins) every time a new release of the KiCAD project is made, by defining a Docker instance in the project repository (containing Linux with KiCAD and relevant plugins and libraries installed). This Docker instance could be automatically activated whenever a Git push is made (validating integrity of all files) and a release is is made (automatically generating documentation and production files).
Obviously this Docker container could also be run client-side for anyone to be able to generate files without having to install and configure KiCAD on the client.

The alternate solution would be to use the KiCAD API directly to ensure installation of the plugin and to activate the functions of the plugin, which I am unaware of is even possible.

Bouni, 2023-01-18:
Good idea!

But takes a lot of work to achive as the GUI is completely entangled with the logic  :expressionless:
I'll concider this as a long term goal but cannot give you any time frame when it will happen. I hope you understand that.

What features do you exactly want to use from the command line? Just the ceration of the files?

hattesen, 2023-01-18:
Generation of files, and preferably checking ERC/DRC beforehand. 

If the structure of the code does not lend itself to be run as a CLI utility, it is unlikely to be worth the investment in time and risk of introducing regression errors to have this feature as a primary goal. But as you say, if you at some point plan to restructure the code anyway, it might be a good idea to try to structure it after the Model/View/Controller design pattern.

Alternatively, would it be possible to control the KiCAD UI including this plugin, from a new/separate plugin, and thereby achieve this automated/batch processing by layering this feature "on top" of the existing plugin, or would that be unrealistic in your view? It would, at least, allow introducing automation features without adding complexity and risk to the existing plugin.

Bouni, 2023-01-19:
After some quick checks it might be possible to generate the fabrication files without the GUI. ERC/DRC is not possible as far as I can tell due to API limitations.

You will defenitely need a KiCAD Installation in your CI container as you need the Python that is shipped with KiCAD, otherwise you'll not have access to `pcbnew` which is defenitely required for this to work.

Also there are some refrences that might need to be changed / monkey patched in order to get this running without GUI but nothing to serious.

PhilippMolitor, 2024-08-16:
I also would love to use this from the CLI.
My use case is quite an important one I think: I want to create a GitHub Actions CI script that automatically creates all needed files for JLC manufacturing, to attach them as artifacts to a release on GitHub.
That way my repo won't be cluttered with output files that shouldn't be there, and you can quickly grab the right version of the files from the releases page.

When building, it would be great to have flags for what to do and what not to do (the stuff from the "settings" UI).
Also it would be great for CI to have a command that allows to only download the database files and then quit (maybe those aren't even needed for CI?)