Bouni/kicad-jlcpcb-tools #268: Double sided placement for JLCPCB Standard

ccdkid, 2023-01-19:
Is there a chance to make it possible?

Bouni, 2023-01-19:
What exactly does not work / is missing for it to work?

I've never used the Standard service so far ...

Bouni, 2023-01-19:
I Just did a quick test and it seems that this as already working fine!

[image]

ccdkid, 2023-01-19:
Sorry, my fault. It works like a charme. Looking forward for the new API. Great work, keep on going 👍

closed 2023-01-19 by ccdkid

bvernoux, 2023-01-19:
> I Just did a quick test and it seems that this as already working fine!
> 
> This 3D view is out of the JLCPCB board viewer: [image]

I confirm I have used with success kicad-jlcpcb-tools (latest version I update it regularly) plugins with KiCad 6.0.10 with JLCPCB with Double Side placement on several boards