Bouni/kicad-jlcpcb-tools #270: Installation not working using plugin manager

filimonic, 2023-01-24:
**Describe the bug**
When adding repo URL to KiCad 6.0.10
22:08:03: Failure when receiving data from the peer
22:08:03: Unable to load repository url

**To Reproduce**
Add repo by url `https://raw.githubusercontent.com/Bouni/bouni-kicad-repository/main/repository.json` 
[image]

# KiCad Version
Application: KiCad (64-bit)

Version: (6.0.10), release build

Libraries:
	wxWidgets 3.2.1
	libcurl/7.86.0-DEV Schannel zlib/1.2.13

Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 19 2022 21:23:04
	wxWidgets: 3.2.1 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.2
	Curl: 7.86.0-DEV
	ngspice: 38
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```
```

Bouni, 2023-01-24:
This is a known issue on Windows machines caused by the used libcurl. This happens for the download of official plugins as well. (The plugins itself, not the repo)

DomoticX, 2023-01-28:
This should be fixed in the KiCad 6.0.11 Release, reading the release notes:

**Windows**
[Use older version of curl that works correctly]

[https://www.kicad.org/blog/2023/01/KiCad-6.0.11-Release/](https://www.kicad.org/blog/2023/01/KiCad-6.0.11-Release/)

Bouni, 2023-01-30:
Fixed in Kicad `6.0.11` and `7.0.0-rc2-62-g971abcfc1a`

closed 2023-01-30 by Bouni