Bouni/kicad-jlcpcb-tools #271: Kicad becomes unresponsive when 'Show part details' clicked

sullivbx, 2023-02-02:
When searching JLCPCB Library, clicking "Show part details" button causes Kicad to become unresponsive.

**To Reproduce**
Steps to reproduce the behavior:
1. Open KiCad project
2. Open PCB Editor
3. Click JLC tools button to open plugin
4. Select a part from the table
5. Click "Select part" button
6. Enter search terms and click "Search" button
7. Select a search result from the list
8. Click "Show part details" button
9. Observe spinning wheel of doom
10. Force quit/restart Kicad

**Expected behavior**
Clicking "Show part details" should show part details window.

# KiCad Version

```
Application: KiCad PCB Editor (64-bit)

Version: (6.0.10), release build

Libraries:
	wxWidgets 3.2.1
	libcurl/7.86.0-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 19 2022 21:23:04
	wxWidgets: 3.2.1 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.2
	Curl: 7.86.0-DEV
	ngspice: 38
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

sullivbx, 2023-02-02:
This may be related to my internet connection and not a problem with the plugin. Connecting through VPN solves this temporarily.

Updating KiCad to 6.0.11 and updating plugin to 2023.01.02 did not fix while VPN disabled.

Bouni, 2023-02-03:
I tried to reproduce on my Win11 machine as well as on my Arch linux machine, on neither this problem shows up.
So I guess its a problem with your internet connection 🤷🏽‍♂️

sullivbx, 2023-02-03:
Thanks for trying, I'll continue to work around it. The plugin is wonderful otherwise.

closed 2023-02-03 by sullivbx