Bouni/kicad-jlcpcb-tools #276: kicad-jlcpcb-tools plugin hangs if project folder is in NFS share

minecraft2048, 2023-02-06:
**Describe the bug**
kicad-jlcpcb-tools plugin hangs if project folder is in NFS share. This is most likely caused by SQLite not playing well with NFS because of bad file locking: https://www.sqlite.org/faq.html#q5

Its using NFSv3
```
n310-drive@n310 ~> sudo mount -t nfs -vvvv 192.168.1.232:/srv /remote
[sudo] password for n310-drive: 
mount.nfs: timeout set for Mon Feb  6 12:25:14 2023
mount.nfs: trying text-based options 'vers=4.2,addr=192.168.1.232,clientaddr=192.168.1.139'
mount.nfs: mount(2): No such file or directory
mount.nfs: trying text-based options 'addr=192.168.1.232'
mount.nfs: prog 100003, trying vers=3, prot=6
mount.nfs: trying 192.168.1.232 prog 100003 vers 3 prot TCP port 2049
mount.nfs: prog 100005, trying vers=3, prot=17
mount.nfs: trying 192.168.1.232 prog 100005 vers 3 prot UDP port 47220

```

If I run kicad using gdb and ctrl-c it when it hangs we can see that it stuck at `fcntl`

```
(gdb) bt
#0  __fcntl64_nocancel_adjusted (fd=19, cmd=6, arg=0x7fffffffb650) at ../sysdeps/unix/sysv/linux/fcntl_nocancel.c:63
#1  0x00007ffff5addee8 in __GI___libc_fcntl64 (fd=<optimised out>, cmd=<optimised out>) at ../sysdeps/unix/sysv/linux/fcntl64.c:51
#2  0x00007fffd135aeb1 in  () at /lib/x86_64-linux-gnu/libsqlite3.so.0
#3  0x00007fffd1369ebb in  () at /lib/x86_64-linux-gnu/libsqlite3.so.0
#4  0x00007fffd136c84d in sqlite3PagerSharedLock () at /lib/x86_64-linux-gnu/libsqlite3.so.0
#5  0x00007fffd12f0930 in sqlite3BtreeBeginTrans () at /lib/x86_64-linux-gnu/libsqlite3.so.0
#6  0x00007fffd1374e11 in sqlite3InitOne () at /lib/x86_64-linux-gnu/libsqlite3.so.0
#7  0x00007fffd1374fe4 in sqlite3Init () at /lib/x86_64-linux-gnu/libsqlite3.so.0
#8  0x00007fffd137502f in sqlite3ReadSchema () at /lib/x86_64-linux-gnu/libsqlite3.so.0
#9  0x00007fffd12fe153 in sqlite3StartTable () at /lib/x86_64-linux-gnu/libsqlite3.so.0
#10 0x00007fffd136eeba in sqlite3Parser () at /lib/x86_64-linux-gnu/libsqlite3.so.0
#11 0x00007fffd139878c in sqlite3RunParser () at /lib/x86_64-linux-gnu/libsqlite3.so.0

```

 I think we can only warn users to not use NFS. Not sure if NFSv4 fixes this

**To Reproduce**
Steps to reproduce the behavior:
1. Make a simple project in kicad on a NFS network share
2. Go to PCB Editor and click JLCPCB tools
3. Kicad hangs

**Expected behavior**
kicad-jlcpcb-tools opens up instead of making kicad hangs

# KiCad Version

```
Application: KiCad

Version: 6.0.10-86aedd382b~118~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.13

Platform: Linux 5.15.0-58-generic x86_64, 64 bit, Little endian, wxGTK, plasmawayland, wayland

Build Info:
	Date: Dec 18 2022 19:39:39
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.86.0
	ngspice: 36
	Compiler: GCC 11.3.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

Bouni, 2023-02-07:
I think in that case you gave yourself the answer. If the problem is with sqlite on an NFS share there's not much I can do about it 🤷🏽‍♂️

minecraft2048, 2023-02-07:
Yeah.. putting this here in case some googled why jlcpcb-tools hangs

Bouni, 2023-02-07:
Yep thats absolutely right to do so. Sorry that I can't do anything to help with this ...

closed 2023-02-07 by Bouni