Bouni/kicad-jlcpcb-tools #285: toolbar buttons too big and weird color

lijon, 2023-02-16:
See screenshot:

[image]

Also the column widths where very long so I had to manually resize each of them to fit the window.

Bouni, 2023-02-17:
Do you use High DPI settings?

lijon, 2023-02-17:
macOS nightly build from yesterday.
It’s an MacBook M1 so yes, high DPI retina screen.

Bouni, 2023-02-20:
Ok I'll take a look at it during this week ...

Bouni, 2023-03-08:
@lijon As I'm on a Windows machine I cant say for sure what the equivalent of high DPI would be on Windows but if I use 200% the Interface looks not as broken as in your screenshot.

[image]

Only the right toolbar is to big ...

I'll see what I can do about that.

DomoticX, 2023-04-24:
This is my screen at 4K 150% zoom, also the toolbar and/or icons are cut-off
[image]

chmorgan, 2025-03-29:
Fixed in newer versions of the plugin.

closed 2025-03-29 by chmorgan