Bouni/kicad-jlcpcb-tools #287: Add support for Kicad

Shackmeister, 2023-02-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Bouni, 2023-02-21:
You mean that parts with the set DNP property shouldn't be added to the list at all?

Shackmeister, 2023-02-22:
Either that or Value set as DNP, and LCSC partnumber removed. Personally I don't care which way

z2amiller, 2026-04-10:
Fixed by: #722 
Fixed by: #723

closed 2026-04-10 by z2amiller