Bouni/kicad-jlcpcb-tools #291: Error: The wx.App object must be created first in Stock Ubuntu 22.04 KiCAD

hansfbaier, 2023-03-02:
The issue and the fix is described in detail here:
https://techoverflow.net/2023/03/01/how-to-fix-kicad-jlcpcb-fabrication-plugin-error-the-wx-app-object-must-be-created-first/

hansfbaier, 2023-03-02:
Seems to be the KiCAD version:
https://forum.kicad.info/t/cant-run-any-installed-plugin-after-update-wx-error/38028
So not a fault of this project.

hansfbaier, 2023-03-02:
The issues persists even after updating to the latest version of KiCAD 6 from the KiCAD ppa.

hansfbaier, 2023-03-02:
[image]

hansfbaier, 2023-03-02:
The fix from the first links works:
```diff
--- a/mainwindow.py
+++ b/mainwindow.py
@@ -64,6 +64,7 @@ ID_CONTEXT_MENU_ADD_ROT_BY_NAME = wx.NewIdRef()
 
 class JLCPCBTools(wx.Dialog):
     def __init__(self, parent):
+        self.app = wx.PySimpleApp()
         wx.Dialog.__init__(
             self,
             parent,
```

Bouni, 2023-03-03:
I need to make sure that this works on all platforms and not just on Ubuntu. If that's not the case we need to bring some sort of OS detection on place in order to get it working on all platforms ...
I guess its just another type of "wrong" wx version in place. wx is a nightmare 😞

Bouni, 2023-03-08:
I've tested your suggestion on Windows 11 and it seems to work.
I'll add this into the next release and hope it doesn't break anything on other platforms ...

closed 2023-03-08 by Bouni

hansfbaier, 2023-03-08:
Great, thanks!