Bouni/kicad-jlcpcb-tools #294: Generate creates an only-header CPL file, no BOM file. Stack trace on stderr. `int - EDA_ANGLE` unsupported.

SmittyHalibut, 2023-03-14:
**Describe the bug**
When generating fabrication files with a board that has components on both top and bottom layers, it generates an CPL file with only a header, and no BOM file.  Bringing up the DEBUG console, I get the following stack trace:

```
Traceback (most recent call last):
  File "/home/mark/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/mainwindow.py", line 894, in generate_fabrication_data
    self.fabrication.generate_cpl()
  File "/home/mark/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/fabrication.py", line 284, in generate_cpl
    self.fix_rotation(fp),
  File "/home/mark/.local/share/kicad/6.0/3rdparty/plugins/com_github_bouni_kicad-jlcpcb-tools/fabrication.py", line 71, in fix_rotation
    rotation = (180 - rotation) % 360
TypeError: unsupported operand type(s) for -: 'int' and 'EDA_ANGLE'
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to PCB Editor
2. Open board that has components on the bottom layer.
3. Open the Python Scripting Console (so you can see the stack trace)
4. Open JLCPCB Tools plugin
5. Click "Generate"
6. Note that the CPL file only has a header, and no BOM is generated.
7. Note the Stack trace.

**Expected behavior**
No stack trace, and fully populated CPL and BOM files.

**More Information**
With some `self.logger.warning()`s, I narrowed down the problem to https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/fabrication.py#L71

Searching back issues, I found a very similar bug report: https://github.com/Bouni/kicad-jlcpcb-tools/issues/107 but it's what resulted in https://github.com/Bouni/kicad-jlcpcb-tools/blob/main/fabrication.py#L63-L70 right above where I'm seeing the problem.

I noticed that my code, and the code in main, both test for `is_nightly` not `= 6.99` as is in the merge for that ticket.  I haven't looked for where that change was made, or why.

My execution path succeeds the `/ 10` branch (confirmed with `self.logger.warn()` output), but fails on `180 -`.  I don't fully understand why that is, but I'm not at expert at this library.

I modified my local code to ALWAYS use `rotation = original.AsDegrees()` (instead of testing on `is_nightly()`) and the problem went away.

I don't fully understand all the version requirements around the use of `.AsDegrees()` vs `/ 10`, but in my case it looks like `.AsDegrees()` is the correct one.  That's why I'm not submitting a Pull Request for the change.  If you'd like me to submit a PR, please explain what conditions can't use `.AsDegrees()` and I'll do the work.  But from where I'm standing, `.AsDegrees()` looks like the right way to do this.

# KiCad Version

JLCPCB Tools version: 2023.03.01

```
Application: KiCad PCB Editor

Version: 6.0.11-2627ca5db0~126~ubuntu22.04.1, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.81.0 OpenSSL/3.0.2 zlib/1.2.11 brotli/1.0.9 zstd/1.4.8 libidn2/2.3.2 libpsl/0.21.0 (+libidn2/2.3.2) libssh/0.9.6/openssl/zlib nghttp2/1.43.0 librtmp/2.3 OpenLDAP/2.5.13

Platform: Linux 5.19.0-35-generic x86_64, 64 bit, Little endian, wxGTK, ubuntu, x11

Build Info:
	Date: Feb  1 2023 20:46:36
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.5.2
	Curl: 7.87.0
	ngspice: 38
	Compiler: GCC 11.3.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

Bouni, 2023-03-20:
Hi,

I tried to reproduce this on my Windows machine but it does not happen to me.
If I put in a `self.logger.error(original.AsDegrees())` I clearly see a `AttributeError: 'float' object has no attribute 'AsDegrees'`

To be honest, I have not the slightest idea why your 6.0.11 seems to have features / breaking changes that were introduced in 6.99 onwards ...

SmittyHalibut, 2023-03-20:
Was `.AsDegrees()` added in 6.99?  Hmm.  Is it possible it was back-ported?  Seems unlikely.

Maybe just try `.AsDegrees()` and catch `AttributeError` and fall back to treating it as a float.

SmittyHalibut, 2023-03-20:
Submitted a PR using that "Try it, fall back" logic.

closed 2023-03-24 by Bouni